Bodmer/TFT_eSPI #558: Text printed to sprite does not wrap

Crushless, 2020-02-26:
Hello,
for some small project (TTGO T-Display as an RSS feed reader), I want to print some changing text, it should be flicker free, and I want to implement smooth vertical scrolling in the future. So I decided to use a sprite as an double buffer, and push it to the display when the text update is done. The text will be multi-line and of unknown length. When I print the text to the display directly, enabling setTextWrap() works as designed, and wraps the text at the end of the display (it would be a nice addition, if the library could be set to wrap only on word boundaries):
[image]
Unfortunately, enabling setTextWrap() on sprites seems to have no effect, so the text gets not wrapped and overflows the display:
[image]
To reproduce this bug (without the RSS and network code), I created a small test Arduino sketch:
[TTGO_TextWrapError.zip](https://github.com/Bodmer/TFT_eSPI/files/4254737/TTGO_TextWrapError.zip).

Bodmer, 2020-02-29:
The Sprite class does not provide a text wrap capability. You need to write a function in the sketch to split sentences at word boundaries. I wrote a function [here](https://github.com/Bodmer/DarkSkyWeather/blob/master/examples/TFT_eSPI_weather/TFT_eSPI_weather.ino#L548) that splits a line into 2 parts at spaces which you could adapt, it does not split words and add hyphens though.

Wrapping text in a Sprite is a potential future enhancement I will think further on.

Bodmer, 2020-03-01:
I had a look at the code and have added the capability to wrap a smooth font in a Sprite. Your example should work now.

closed 2020-03-01 by Bodmer

Bodmer, 2020-03-01:
I found this function in one of my sketches which I have adapted so you can add to your example sketch:
```
void printSplitString(String text)
{
  int wordStart = 0;
  int wordEnd = 0;
  while ( (text.indexOf(' ', wordStart) >= 0) && ( wordStart <= text.length())) {
    wordEnd = text.indexOf(' ', wordStart + 1);
    uint16_t len = buffer.textWidth(text.substring(wordStart, wordEnd));
    if (buffer.getCursorX() + len >= buffer.width()) {
      buffer.println();
      wordStart++;
    }
    buffer.print(text.substring(wordStart, wordEnd));
    wordStart = wordEnd;
  }
}
```

Example usage:
```
const String ExampleText PROGMEM = "The quick brown fox jumps over the lazy dog. Jackdaws love my big sphinx of quart. Pack my box with five dozen liquor jugs.";

printSplitString("Sprite:\n" + ExampleText);
```