Bodmer/TFT_eSPI #559: crash enc28j60 with TFT_eSPI

shalom198, 2020-02-27:
hello , i'm using the uipethernet.h lib with the enc28j60 for my project with mini D1 &  tft 1.8 7735 and works fine
know i replace the tft with tuch and lib TFT_eSPI and the ethernet cant connect 
the screen works fine and the tuch works fine when i connecting with the wifi 
but , when i try to connect with the ethernet its fail ,no connection 
when i chang only in the program to lib of 7735 the enc28j60 works fine 

what can it be ? i think somthing in the SPI or CS is conflict, but didnt find what ?
plz help

Bodmer, 2020-02-29:
List all the connections between the ESP32, TFT and the EN28J60 so I can review possible causes. Provide a link to the EN28J60 library you are using as the problem may be in that library.

shalom198, 2020-02-29:
i using this lib https://github.com/UIPEthernet/UIPEthernet
the conennctions are :
mini D1       
D5             sck
D6            miso
D7            mosi

D2            cs     - screen 
D3            cs     - enc28j60
D4            cs     - tuch

I would like to mention that all connections remain the same 
and i am changing only the screen to somthing else the ethernet works fine 
and if i change only the ehternet to wifi conection the tuchscreen works fine 
with the same connection and configurtion at the program

shalom198, 2020-02-29:
i change in the user.h the cs from D2 to D4 because the D2 is used for the screen in my program

shalom198, 2020-02-29:
#define TFT_CS   PIN_D2  // Chip select control pin D8
#define TFT_DC   PIN_D1  // Data Command control pin
#define TFT_RST  PIN_D0  // Reset pin (could connect to NodeMCU RST, see next line)
#define TOUCH_CS PIN_D4     // Chip select pin (T_CS) of touch screen

thats the configurtion in my sketch 

i tryd to change my program like this

//#define TFT_CS   PIN_D8  // Chip select control pin D8
//#define TFT_DC   PIN_D3  // Data Command control pin
//#define TFT_RST  PIN_D4  // Reset pin (could connect to NodeMCU RST, see next line)
//#define TOUCH_CS PIN_D2     // Chip select pin (T_CS) of touch screen

but this didnt work as the other

Bodmer, 2020-02-29:
Have you enabled transaction support by adding this line to your settup file:

```
#define SUPPORT_TRANSACTIONS
```

shalom198, 2020-03-01:
no 
wher to add it

shalom198, 2020-03-01:
i found where it is 
and its work fine 
thanks alot

Bodmer, 2020-03-01:
Great!

closed 2020-03-01 by Bodmer

shalom198, 2020-03-01:
thanks alot😊

‫בתאריך יום א׳, 1 במרץ 2020 ב-18:56 מאת ‪Bodmer‬‏ <‪notifications@github.com
‬‏>:‬