Bodmer/TFT_eSPI #560: ILI9481 SPI ESP32 stripes on screen - Defective LCD screen ?

Puebllo, 2020-02-27:
Hello Bodmer,
I've made a PCB, which translates signals for LCD from SPI to D0:D15, based on this schematic:
https://cloud.githubusercontent.com/assets/3693437/4249610/2d25eed4-3a79-11e4-8418-b8439db1cc4f.png

LCD works, but there are stripes on it when I try to display something. Doesn't matter if it's static (text) or dynamic (3D cube wireframe animation) content. I also have original Waveshare 3,5" Rev2.0 screen and when I plug Waveshare screen into my pcb it works perfectly, but when I plug my LCD to Waveshare PCB it have stripes aswell. My LCD only works with setup for Waveshare screen (setup 11).

Link to video (horizontal lines):
https://youtu.be/x-ap-YBXuRU

And this starts me to wondering- is my LCD poor quality, or it's something related with library.
The big difference (I think) is, that the Waveshare screen is on ILI9486 driver.

Here is the list what I've tried:

- Using user setup 11 and 20
- Making own user setup with drivers: 9481, 9486 and 9488 and changing spi frequencies in range from 10MHz to 27MHz (1MHz increment)
- Looks like drivers 9481, 9488 don't work at all (black screen)

Enviroment: 
- Arduino IDE version: 1.8.9
- Board: ESP32 Dev Module
- tft_eSPI newest version
- IM[0:2] set to: 0,1,0

If it will be LCD related problem, could you recommend some bare 3,5" IPS LCD screen ?

Puebllo, 2020-02-27:
Okay,
I think I've figured that out. I'm using setup 11 for raspberry pi and in ILI9486_Init.h after 0x11 command I added commands for power settings:

    writecommand(0xD0);
    writedata(0x07);
    writedata(0x42);
    writedata(0x17);

    writecommand(0xD1);
    writedata(0x00);
    writedata(0x07);
    writedata(0x10);

    writecommand(0xD2);
    writedata(0x01);
    writedata(0x02);
	
    writecommand(0xC0);
    writedata(0x10);
    writedata(0x3B);
    writedata(0x00);
    writedata(0x02);
    writedata(0x11);
	

and value of 3rd parametr for command 0xD0 changed from 0x18 to 0x17. I'm guessing it will be now fine tuning settings (sometimes lines appears one white background but they're almost invisible).
I'll try to make standalone setup or fix already existing for this lcd.

Bodmer, 2020-02-29:
The circuit you are using introduces a unique need to send SPI commands as 16 bits, whereas normally they would be 8 bits. The circuit you have used is specific to RPi displays and is constrained in the maximum SPI clock frequency to circa 20MHz.  These factors mean performance is typically lower than a normal SPI or 8 bit parallel interface. To force 16 bit commands the following setup should work without needing to change the init code:
'''
#define RPI_DISPLAY_TYPE
#define ILI9481_DRIVER
'''

Puebllo, 2020-02-29:
Thank You for your explantion. Now I understand. After mentioned change it works with ili9481 driver, but still 3rd parameter in 0xD0 need to be changed to 0x17 - I'm guessing it's dependend on lcd manufacturer

Bodmer, 2020-03-01:
You may find the commenting out all 0xD0, 0xD1 and 0xD2 settings and allowing the defaults to be used gives a better colour balance.

closed 2020-03-01 by Bodmer

Puebllo, 2020-03-01:
You're right it helped with color, but issue with stripes came back. Have you any ideas how to fix this ? 

Here's picture:

https://imgur.com/a/mwA0blC

Bodmer, 2020-03-01:
Try putting in just the 0xD0 command, maybe that is required for your display.

Puebllo, 2020-03-01:
Yep 0xD0 does the trick, but I had to tweak it. It's not perfect, but it needs now fine polishing :)

dexter323i, 2021-07-17:
Hi @Puebllo,
Did you succeed in fine polishing? :) I have EXACTLY the same striping issue like you had.  The command:
writecommand(0xD0);
writedata(0x07);
writedata(0x42);
writedata(0x17);
Also removed my stripes, but colors seem a bit off. Could you share your final commands and parameters? I would appreciate it so much!

Puebllo, 2021-07-18:
Hi @dexter323i,
unfortunately, I've dropped idea of using this lcd in parallel mode (too much gpios are needed and also circuit is much complex). I'm using it in SPI mode. With colors I think i didn't done much more than what I described here. Changing INIT modes for drivers, at beginning of ILI9481_Init.h wont change much because in parallel you're forced to use 16bit pallete color.

https://github.com/Bodmer/TFT_eSPI/blob/58f457ba9703613bfcd70b411d66bbbc948fa815/TFT_Drivers/ILI9481_Init.h

Keep in mind, that in parallel mode you can use only 16bit color pallette (RGB565). If you put 24bit color in code as hex, then it'll be completely different than you wanted. It needs to be in 16bit format. You can get colors from this site:
http://www.barth-dev.de/online/rgb565-color-picker/

Also I think, that in SPI mode colors looks much better, because it uses 18bit color pallete

dexter323i, 2021-07-18:
Hi @Puebllo,
Thanks for your quick response! Actually, I came from Nextion world, so I got used to the RGB565 color palette. Yes, the parallel interface eats up many GPIO pins, but thank God, I have enough.
With the colors you are right, but the speed increase is HUGE. I need quick refresh in my project, so I use sprites to store images in RAM. Refreshing the whole 480x320 screen from an already loaded sprite takes around 220ms to complete in SPI mode! Because it can only handle 15MHz SPI speed. With short cables. This is a disaster for me! With this 8bit parallel interface, the same process takes ~67ms. This is a game changer difference!

For future visitors: I finally ended up using the default **ILI9481_Init.h** driver, with the default selected **#define ILI9481_INIT_1** . I only changed the the 0xD0 command to this:
    writecommand(0xD0);
    writedata(0x07);
    writedata(0x41);
    writedata(0x17);

I have no stripes now on the screen, with good enough colors.