Bodmer/TFT_eSPI #561: Feature Request - multiple displays

ZL1CVD, 2020-03-01:
I wish to use 3 or 4 displays. Via hardware, each display will be allocated a unique port for CS. What is the best way or facilitate this? User_Setup.h only has provision for 1x CS.

Bodmer, 2020-03-01:
This is not a feature I will be adding. The chip selects to the different displays can be managed in your sketch.

In the setup file comment out the line that defines TFT_CS then the library will not control any pin for the chip select. In the setup() you must set all TFT CS lines low before the tft.init(); then set high

To write to just one display then just set one CS line low.

If you want to clear all screens at the same time set all CS lines low and do a tft.fillScreen();

If you wish to read from the TFT CGRAM then clearly you must only select one TFT at any one time.

Note that if you set the cursor position etc for one display then you will need to set it again for a different display.

closed 2020-03-01 by Bodmer

ZL1CVD, 2020-03-02:
Perfect!
Thank you!