Bodmer/TFT_eSPI #562: Not able to run on stm32f103vxxx Arduino_Core_STM32

gsaikiran29, 2020-03-02:
hello,
I am using ili9341, [https://github.com/stm32duino/Arduino_Core_STM32](url) core, Vet6 controller.
I have added pins in User_Setup.h
```#define TFT_MISO PB14
#define TFT_MOSI PB15
#define TFT_SCLK PB13
#define TFT_CS   PB12
#define TFT_DC    PD9
#define TFT_RST   PD8
```
my tft is on spi2.
I am using utft demo code and added this SPI initialization in setup()
```SPI.setMISO(PB14);
SPI.setMOSI(PB15);
SPI.setSCLK(PB13);
SPI.setSSEL(PB12);
```
compiled and successfully uploaded but display is blinking in white color.

Bodmer, 2020-03-02:
The library only supports use of the STM32F103 SPI 1 and DMA on port/pins PA5-7. See [setup 32](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup32_ILI9341_STM32F103.h).

gsaikiran29, 2020-03-03:
thank you for the response @Bodmer 
I added my pins in Setup32_ILI9341_STM32F103.h
its came. but printing is much slower.
utft demo code gives (16240msec) in result.

gsaikiran29, 2020-03-03:
I want use custom fonts like sumotoy's tft_ili9341 library he provided font and image converter application, but That library doesn't have updated features like string alignment.

Bodmer, 2020-03-03:
I suspect your setup is still not correct for the STM32 and the slower generic driver is being used. It is always best to use a tried and tested setup before embarking on a bespoke setup.

You must use the [#define to invoke the optimised SPI drivers](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup32_ILI9341_STM32F103.h#L7-L8). The time for the UTFT demo should then be circa 3800ms i.e. 4 times faster.

I am not familiar with sumotoy's library but I expect it uses a bespoke format not supported by TFT_eSPI. TFT_eSPI supports Adafruit_GFX style fonts which can be generated using online tools. It also supports smooth (anti-aliased) fonts that can be generated from your computers font files. See examples.

closed 2020-03-03 by Bodmer