Bodmer/TFT_eSPI #563: Examples/Keypad_###x### display issue

cmidgley, 2020-03-03:
In the examples folder, for both 320x240 and 480x320 there appears to be a small bug.  The result is that when pressing a keypad button, a large rectangle clears out a portion of the page.  In this case, it is after pressing the "5" button once:

[image]

It appears the "status" method is setting "tft.setTextPadding(240)" but then later when calling "drawButton" this padding state remains and results in the large rectangle.  It can be fixed by resetting padding via "tft.setTextPadding(0)" prior to the "drawButton" calls.

I'm new to TFT_eSPI (as in, just started using it today), so I wasn't sure if this was a bug in the sample code, or perhaps it was something that drawButton should be handling?  But I *think* it isn't my hardware setup...?

Bodmer, 2020-03-03:
The button class is based on the AdafruitGFX class and is very basic, it over-rides the text colours, text size and text padding so you need to reset them to your desired values after creating the buttons.

A potential future enhancement is for the class to restore old values.

closed 2020-03-03 by Bodmer

cmidgley, 2020-03-03:
Makes sense.   I think that means that the examples/320 x 240/Keypad_320x240/Keypad_320x240.ino file (and the it's brother under 480 x 320) have a bug as they need the tft.setTetPadding(0) at line 127 (or have the status function set it back to 0, before line 281) in order to render the buttons correctly.  

Thanks for all your hard work on this library.  Made it possible to go from zero to working status display for my project in one day!