Bodmer/TFT_eSPI #564: printLn not working

TigerClaw33, 2020-03-03:
Hi,

your lib works fine for my Gauge application but I would like to use some functions from the original adafruit lib but this would not be work ...

My problem is that commands like tft.begin() are not working and println is also not working, I could use tft.drawCentreString insteadt of tft.prinln but if I create a string and put it as a header with a background I dont see the background. I see something like repetitions from the font with black background.So I would use some functions from basic lib but there are not working

Bodmer, 2020-03-03:
I do not understand the issue with tft.begin();  It is used in all the sketches and must have been in your working Gauge application.

println works the same as the Adafruit library as far as I am aware. Provide an example sketch and explain what is not working with reference to that sketch. Test it also with the Adafruit_GFX library and explain the differences you get.

TigerClaw33, 2020-03-03:
Hello, 

if I try to use both libs in the same scatch I get an error:

`#include <Adafruit_GFX_AS.h>     // Core graphics library
#include <Adafruit_ILI9341_AS.h> // Fast hardware-specific library
#include <Adafruit_GFX.h>     // Core graphics library
#include <Adafruit_ILI9341.h> // Fast hardware-specific library

Adafruit_ILI9341_AS tft = Adafruit_ILI9341_AS(cs, dc, rst);
Adafruit_ILI9341 tft = Adafruit_ILI9341(cs, dc, rst); 
`
I wanted to use the original lib to to use commands like println because if i use println with your lib i don't see any chars. Only drawString() works with your library.

And I notticed that your lib uses tft.init and the original one tft.begin. Is there a reason why?

Bodmer, 2020-03-03:
You seem to be getting your libraries muddled up.
Look at [this example](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/160%20x%20128/TFT_Print_Test/TFT_Print_Test.ino).
tft.init() and tft.begin() are equivalent and are for backwards compatibility.

TigerClaw33, 2020-03-03:
Also working with your Gauge V2 lib for Arduino Uno?

Bodmer, 2020-03-03:
Guage V2 lib? Not aware of that.

Bodmer, 2020-03-04:
I am closing this as there appears to be confusion of this library with another. Start by looking at the examples if you wish to use TFT_eSPI.

closed 2020-03-04 by Bodmer