Bodmer/TFT_eSPI #565: Feature Request - 8 bit parallel mode for ESP8266 in User_Setup.h

Derek-K, 2020-03-03:
After reading most of the code, it seems like 8bit parallel mode only available for ESP32. Limited I/O on ESP8266 could be one of the reasons. However, I have tested ESP8266-12 indeed has 11 useable output pins.

May I propose, D0-D7 for the parallel interface.
D8, Rx (D9) & Tx (D10) for DC, CS & WR.

And just hope power on reset of LCD works (or can it tie to ESP8266-12 RST pin so both reset at the same time?) :)  No read from LCD memory is needed.

I have read TFT_eSPI_ESP32, wonder how easy/difficult to port the parallel code over to TFT_eSPI_ESP8266... Perhaps you can test it with the parallel mode ILI9341?

Thanks

Bodmer, 2020-03-03:
The 8 bit parallel interface capability for the ESP32, STM32 and generic Arduino boards is supported but not for ESP8266 as you have surmised.  I will not be adding parallel support for the ESP8266 as the SPI interface speeds are very good due to the hardware buffering, so parallel mode is likely to be slower.

If you wish to experiment then the simplest route is to disable the ESP8266 specific driver and force use of the generic driver (which already supports 8 bit parallel) by changing [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L49) to:
```
#elif defined (xxxxxESP8266)
```

closed 2020-03-03 by Bodmer

Bodmer, 2020-03-03:
Update. I see the parallel mode is not yet fully implemented for the generic driver so my suggestion for experimentation will not work. Sorry!

Derek-K, 2020-03-03:
> 
> If you wish to experiment then the simplest route is to disable the ESP8266 specific driver and force use of the generic driver (which already supports 8 bit parallel) by changing [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_eSPI.h#L49) to:
> 
> ```
> #elif defined (xxxxxESP8266)
> ```

OK, this part I get. I can make my hybrid Processors/TFT_eSPI_xxxESP8266.h with parallel support borrow from ESP32 code (or write my own). This makes sense...

> Update. I see the parallel mode is not yet fully implemented for the generic driver so my suggestion for experimentation will not work. Sorry!

Not sure what you mean? Say if I implemented the 'parallel mode' above, wouldn't all low level communication to LCD will select either SPI or Parallel Mode just like using ESP32? Or there is still something missing I should be aware of?

BTW, debug is going to be painful, since I will be using Tx/Rx line to control the LCD as well... 🙄

Bodmer, 2020-03-04:
The ESP32 and STM32 implementation will give you clues on the implementation of an 8 bit interface but the code is not directly usable.  The easiest route by far will be to use an ESP32 in parallel mode or an ESP8266 with SPI.