Bodmer/TFT_eSPI #566: Since library update on 02nd March 2020 colours all wrong drawing JPEG or BMP from SPIFFS to ILI9488 TFT with ESP8266

mattstroud81, 2020-03-03:
Hi,

I have been using an ILI9488 TFT with an ESP8266 Wemos D1 Mini, i had a go at loading a custom image to SPIFFS and using the example was able to display this on the TFT and it looked absolutely fantastic! I was able to do the same with both JPEG and BMP images using parts from both examples.

Soon after this the IDE told me a library update was available, it was for TFT_eSPI so i updated it. Sometime later, maybe 30 minutes or so?? all of a sudden the communication with the TFT stopped working. In realised that this was because the library update had reset the link to my custom setup select file, i corrected this line and re-sent the programme to the ESP8266 again and the image was drawing on the screen but all the colours were incorrect and the picture looks all wrong. It is doing the same with both JPEG and BMP images stored in SPIFFS. The example which stores the image data in a separate '.h' file works and colours are normal, so it looks like this issue is related to SPIFFS specifically.

Can this please be looked into, i'm working on a birthday present for my brother, but i can't continue now until i can figure out how to fix this issue.

Many thanks
Matt

mattstroud81, 2020-03-03:
Ok, i have done a little testing and have a very interesting finding! I tested a simple screen fill by doing:

tft.fillScreen(TFT_RED)  -->  Screen was Red
tft.fillScreen(TFT_GREEN)  -->  Screen was Green
tft.fillScreen(TFT_BLUE)  -->  Screen was Blue

So that means the screen is ok, communication pins etc must all be ok, library is working as expected here.

I then created three 320 x 480 Jpeg images, one completely red (called 'Red.jpg') and the others, Green and Blue.

I then performed the following:

drawJpeg("/Red.jpg", 0, 0)  -->  Screen was BLUE!!
drawJpeg("/Green.jpg", 0, 0)  -->  Screen was RED!!
drawJpeg("/Blue.jpg", 0, 0)  -->  Screen was GREEN!!

So clearly something is happening here when reading image data from the SPIFFS or rather when drawing the data to the screen, is it drawing the RGB colour data in the wrong order? Has this been changed in the library on this latest update? As it was working normally just before the update.

mattstroud81, 2020-03-03:
Ok, the plot thickens! I just clicked on the text 'STM32 Update' next to bodmers name on the '<> Code' tab and the first thing i see is (attached picture) 

[image]

I know this relates to the STM32, but it clearly looks like it is related to colour channels, it looks more and more like the latest update may have 'broken' the JPEG and BMP image functions. Or is there something else in need to add in my code to allow for this in order to get images loading from, SPIFFS to display properly again?

Any help would be really very much appreciated. @Bodmer - I'm guessing you are the best person to know what was changed in this latest update? Many thanks for all your help and for your absolutely fantastic library and all your hard work in the first place :-D

mattstroud81, 2020-03-04:
To show exactly what i'm seeing when uploading an image, here is the 'lena20k.jpg' file that shows in one of the examples. [image]

When i run this on my ESP8266 with ILI9488 screen this is the image i get displayed: 
[image]

Can anyone offer any assistance as i'm not quite sure where to go or what to try next?

Bodmer, 2020-03-04:
Looks like there is an endianess inconsistency.

Try adding this line after tft.begin() in the sketch setup():

```
tft.setSwapBytes(true);
```

mattstroud81, 2020-03-04:
Oh wow! and now it is completely back to normal! absolutely brilliant, thank you so much for such a quick response :-) and thank you for all your amazing work with these libraries. I'm very new to Arduino and with the fantastic libraries you have created i am having a heck of a lot of fun and some real WOW momets :-) Thank you so much.

mattstroud81, 2020-03-04:
Ok, now i'm really confused! I added the line you mentioned and straight away the pictures looked perfect! Just changed the file names to the pictures of my girls and now they are loading with the same graphical mess as before??

I changed it to tft.setSwapBytes(false); as well as true and removed the line altogether and all three times they are still loading with the wrong blocky messy colours. What on earth is going on? i'm so confused.

mattstroud81, 2020-03-04:
@Bodmer - So it turns out, adding the tft.setSwapBytes(true); fixes the problem when loading Jpeg files, but does not fix the problem for loading BMP files.

So it looks like i will continue this project using jpeg files then :-) once again thank you for your help and all your work. and fantastic support, very much appreciated.

Bodmer, 2020-03-04:
[This line over-rides the setting  in setup()](https://github.com/Bodmer/TFT_eSPI/blob/master/examples/Generic/TFT_SPIFFS_BMP/BMP_functions.ino#L38), so change that line to false until I resolve the inconsistency.

mattstroud81, 2020-03-04:
Ah fantastic, thank you. I will alter that line for now then after work. Many thanks again.

Bodmer, 2020-03-07:
Should be fixed not so you can revert the temporary fixes.

closed 2020-03-07 by Bodmer

mattstroud81, 2020-03-08:
Hi Bodmer, just wanted to say, thank you again for the fantastic library and fantastic support, i have reverted to the updated library release and all is working perfectly. Also thank you for the speed increase in drawing full screen images :-) The JPEG drawing speed has been reduced by half! very impressed! thank you so much :-D

commit ac8845d5 (2020-03-07): Fix #566 plus others

Fix image rendering issue.
Deprecate use of pushColors.
Improve ES8266 image rendering performance for ESP8266 and ILI9488.
Add getTextPadding().