Bodmer/TFT_eSPI #567: Work this on a Heltec LORA V2

cpunkt1103, 2020-03-06:
Hello.
I try to connect a ILI9431 TFT to my Heltec LORA V2 Modul. I have big Problems with the TFT
is will not work.... 

I use the onbord Lora to send/recive data, and i connectet the TFT to the Mosi/Miso pin and then
i connectet the cs and dc pins to diverent pins and defined in the sketch, nothing work..

The TFT coneccted to a ESP32Vroom work fine.... inc Sd-Card ... 

Mayby you have a idea , what is the problems resolve...

thx

Bodmer, 2020-03-06:
How have you connected the Heltec module to the TFT?
Provide a link to the TFT vendors web page, make sure the back of the TFT module on the web page looks exactly like yours.
Provide web links to the Heltec module vendor page, datasheet and ideally the schematic (citcuit diagram).
Provide a copy of your TFT_eSPI setup file.

cpunkt1103, 2020-03-07:
Hello ... her is my report ...
The TFT with SD is this SPI / ILI9341 
https://www.ebay.de/itm/322625783359

The Heltec ESP 32 V2 LORA is this
https://www.passion-radio.com/tracker/lora-esp32-heltec-883.html 
and on this site you find the shematic ...

Now the Part of the User_Setup.h from your eSPI.

Line 19 : #define ILI9341_DRIVER
Line 161 to 169 im chanced to the Heltec Pinout
// For ESP32 Dev board (only tested with ILI9341 display)
// The hardware SPI can be mapped to any pins
//Change for the Heltec Test
#define TFT_MISO 19
#define TFT_MOSI 27
#define TFT_SCLK 5
#define TFT_CS   25  // Chip select control pin
#define TFT_DC   23  // Data Command control pin
#define TFT_RST   13  // Reset pin (could connect to RST pin)

I use the Heltec shematic to connect to the SPI pins with is connectet the LORA. 
The Lora works fine (I used this for my Projekt) with the libary from Heltec.

When i take this to my ESP32Vroom with the Adafruit.ILI9341 Lib its work fine... and the SD is usable..

I have testet the Display wit our eSPI on the ESP43Vroom it wil even not work .. i testing make noe a new test.

Thx for possible Help ...MFG Christian from Germany 
now ...

Bodmer, 2020-03-07:
Run the "Read_User_Setup" example to make sure the setup is being picked up by the compiler.

Run a TFT_eSPI example but make sure you set GPIO 18 high (disables LORA chip select) before calling tft.begin();

Try a different pin to GPIO23 for TFT DC.

Make sure the LED pin (GPIO 25) is not being toggled by the Lora library as that will reset the TFT configuration. Try a different pin for reset.

cpunkt1103, 2020-03-07:
Hello....
I do not understand .... i have disconnected my display and make the wires new to the display ...
and his running now.... thanks a lot ... i try later the SD on the diplaybord and the LORA on the Heltec ...
i bring back the results ... mayby for later/other problems ....

Thanks from Germany ..... and have a nice day ....

PS.: your eSPI  is mayby fine for my Projekt... i try the Scroll-Text and the Graphic-Sprites for Diagramms and Bars ....

closed 2020-03-08 by Bodmer