Bodmer/TFT_eSPI #572: printToSprite doesn't print first character (Demo 4).

Gnitry, 2020-03-19:
2.2 TFT SPI 240x320. ILI9341 is used.

Steps to reproduce:
* Open examples\Smooth Fonts\FLASH_Array\Font_Demo_4_Array\Font_Demo_4_Array.ino
* Change line 
`spr.printToSprite("Small 15pt font");    // Prints to tft cursor position, tft cursor NOT moved`
to 
`spr.printToSprite("AA");    // Prints to tft cursor position, tft cursor NOT moved`
* Run
* The display will show only one character "A".

If I try to print "ABC", it will be printed correctly.
If I try to print "A", nothing will be printed.

Test project (using PlatformIO) is attached:
[testespi.zip](https://github.com/Bodmer/TFT_eSPI/files/4356157/testespi.zip)

Bodmer, 2020-03-19:
I have updated the Sprite.cpp file to fix this.
Thanks for reporting this issue.

closed 2020-03-19 by Bodmer

NYFB8888, 2023-12-30:
I have a similar problem using V2.5.34.
Sp.printToSprite(text) where text is a text string = "0" gives nothing, but spr.printToSprite("0"+text) gives me 00 as required.
Moreover, spr.printToSprite(" "+text) does not display any text.

I think that
    while (n < len)
    {
      uint16_t unicode = decodeUTF8((uint8_t*)cbuffer, &n, len - n);
      //Serial.print("Decoded Unicode = 0x");Serial.println(unicode,HEX);
      //Serial.print("n = ");Serial.println(n);
      drawGlyf(Unicode);
    }
DrawGlyph does not print spaces.

commit bdf2c9ba (2020-03-19): Fix #572