Bodmer/TFT_eSPI #574: Lolin32 ( ESP32) with MRB3205

kpaul1978, 2020-03-20:
Hello and many thanks in advance for your help. 
I want to connect this : http://www.lcdwiki.com/3.2inch_16BIT_Module_ILI9341_SKU:MRB3205#Interface_Definition

with the Lolin32 board ( based on ESP32)

https://arduino-projekte.info/wp-content/uploads/2017/07/lolin32_pinout.png

I tried different configurations and pin assignment but nothing worked so far. ( It must be in parallel mode)
Any help would be greatly appreciated, especially in these times when working from home is a must :)

Paul

Bodmer, 2020-03-22:
The library supports 8 bit displays only so you need to modify the display PCB as detailed in your link.

Setup 14 should work, [see here.](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup14_ILI9341_Parallel.h#L9-L24).  Comment out [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L22), and un-comment [this line](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L38).

Note the comments and be aware that some ESP32 pins are input only (i.e. cannot be used as an output to drive the display.

kpaul1978, 2020-03-23:
Hello and many thanks for your answer. 
I managed to solve it and it works perfectly. 

Thank you so much!

closed 2020-03-23 by kpaul1978