Bodmer/TFT_eSPI #575: Esp8266 custom fonts size

alex920a, 2020-03-21:
Hello! i'm having issues with fonts bigger than 72 points. I've generated the fonts via : 
https://rop.nl/truetype2gfx/.
With fonts with points < 72 , no issues at all. 
How Can I solve this?

Bodmer, 2020-03-22:
This is due to a limitation of the font format that Adafruit designed. It uses a 16bit offset pointer so this means that large fonts cause the pointer to overflow.  For example here is the en of the offset table in the font file for a 72 point front from the website you quoted:
```
  { 64295,  65,  74,  71,    2,  -73 },   // 0x78 'x'
  { 64897,  64, 105,  71,    3,  -73 },   // 0x79 'y'
  {   201,  60,  74,  71,    4,  -73 },   // 0x7A 'z'
  {   756,  33, 133,  47,    6, -102 },   // 0x7B '{'
  {  1305,   9, 133,  37,   14, -102 },   // 0x7C '|'
  {  1455,  33, 133,  47,    4, -102 },   // 0x7D '}'
  {  2004,  61,  24,  82,   11,  -61 } }; // 0x7E '~'
```

Notice how the first column wraps at the 16 bit pointer limit and so the 'z' character has the wrong offset. Unlike Adafruit_GFX, the TFT_eSPI library can handle a 32bit offset pointer but the font generator link uses the expected 16 bit limit.  You have the following options:

1. You need to find a converter that does not roll over
2. Find a font generator where you can limit the font file size by reducing the character set range
3. Manually add 65536 to all the rolled over pointers, for example the z pointer above should be 201+65536 = 65737
4. Use a different font format, e.g. smooth font with TFT_eSPI.

Once you have fixed the pointer issue you then have another limitation imposed by the Adafruit format which is a 8 bit signed offset which limits the font size, probably to 127 pixels above or below the font baseline.

Bodmer, 2020-03-22:
Experiments indicate the 8 bit signed value for the font datum reference limits the font size to a maximum of around 90 points (FreeSans used for this test). Again this is a limit imposed by the Adafruit_GFX format.

alex920a, 2020-03-23:
So How can I fix this? How can enlarge the custom font to march the tft size? Cordiali saluti,Alessandro Di CesareTeam MarduM. +39 3342274969infomarduproductions@gmail.comP.IVA 03789591207 | Intestatario Alessandro Di CesareWebsite | Facebook | Instagram  Da: BodmerInviato: lunedì 23 marzo 2020 00:21A: Bodmer/TFT_eSPICc: Alessandro Di Cesare; AuthorOggetto: Re: [Bodmer/TFT_eSPI] Esp8266 custom fonts size st7789 240x240 (#575) Experiments indicate the 8 bit signed value for the font datum reference limits the font size to a maximum of around 90 points (FreeSans used for this test). Again this is a limit imposed by the Adafruit_GFX format.

alex920a, 2020-03-23:
Thank you! I will try! Do you know any good font converter to solve this? Cordiali saluti,Alessandro Di CesareTeam MarduM. +39 3342274969infomarduproductions@gmail.comP.IVA 03789591207 | Intestatario Alessandro Di CesareWebsite | Facebook | Instagram  Da: BodmerInviato: domenica 22 marzo 2020 22:59A: Bodmer/TFT_eSPICc: Alessandro Di Cesare; AuthorOggetto: Re: [Bodmer/TFT_eSPI] Esp8266 custom fonts size st7789 240x240 (#575) This is due to a limitation of the font format that Adafruit designed. It uses a 16bit offset pointer so this means that large fonts cause the pointer to overflow. For example here is the en of the offset table in the font file for a 72 point front from the website you quoted:  { 64295,  65,  74,  71,    2,  -73 },   // 0x78 'x'  { 64897,  64, 105,  71,    3,  -73 },   // 0x79 'y'  {   201,  60,  74,  71,    4,  -73 },   // 0x7A 'z'  {   756,  33, 133,  47,    6, -102 },   // 0x7B '{'  {  1305,   9, 133,  37,   14, -102 },   // 0x7C '|'  {  1455,  33, 133,  47,    4, -102 },   // 0x7D '}'  {  2004,  61,  24,  82,   11,  -61 } }; // 0x7E '~'Notice how the first column wraps at the 16 bit pointer limit and so the 'z' character has the wrong offset. Unlike Adafruit_GFX, the TFT_eSPI library can handle a 32bit offset pointer but the font generator link uses the expected 16 bit limit. You have the following options:You need to find a converter that does not roll overFind a font generator where you can limit the font file size by reducing the character set rangeManually add 65536 to all the rolled over pointers, for example the z pointer above should be 201+65536 = 65737Use a different font format, e.g. smooth font with TFT_eSPI.Once you have fixed the pointer issue you then have another limitation imposed by the Adafruit format which is a 8 bit signed offset which limits the font size, probably to 127 pixels above or below the font baseline.

alex920a, 2020-03-23:
Thank you! I will try! Do you know any good font converter to solve this?

alex920a, 2020-03-23:
> Experiments indicate the 8 bit signed value for the font datum reference limits the font size to a maximum of around 90 points (FreeSans used for this test). Again this is a limit imposed by the Adafruit_GFX format.

So, instead of using the adafruit gfx, can I use the smooth font and size it up to the height of the tft?

alex920a, 2020-03-23:
I've solve this using as you said the smooth font (thanks for the processing sketch). 
Just last question: how can i reduce the space between characters?

Bodmer, 2020-03-23:
Adjusting the spacing between characters is called kerning. The distance between characters is set by the source font file and there is no built in function to change the spacing for individual characters or the whole font.

It is possible to adjust the spacing for all characters [by changing this line](https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Smooth_font.cpp#L193) which sets the cursor advance for the next character, but that would then affect every font loaded.  As the font metrics are kept in memory it would be possible to write a sketch function to adjust the value.

closed 2020-03-23 by Bodmer