Bodmer/TFT_eSPI #578: Only balck screen on TTGO T-Display ESP32 1.14inch LCD

BojanJurca, 2020-03-22:
Hi, I've got TTGO T-Display ESP32 WiFi Bluetooth 1.14 inch LCD that only displays black screen. Example says DON'T FORGET TO UPDATE THE User_Setup.h FILE IN THE LIBRARY but I can't find it anywhere in Arduino folder. Any idea?

liuzewen2, 2020-03-22:
你好，这个要在我的电脑->文档->Arduino->libraries->TFT_eSPI-master  目录下，即可看到user_Setup.h文件。

BojanJurca, 2020-03-22:
Thank you, but the sketch compiled OK previously without libraries being there. I downloaded complete project anyway ant put it into ...\libraries folder, modyfied the files and compiled cketch again. Compiler is not taking my modyfied files but keeps taking original ones from somewhere else.

Bodmer, 2020-03-22:
Check you User_Setup_Select.h file inside the library is calling up the correct setup file. Read the comments, make sure **only one** setup file is called up.

liuzewen2, 2020-03-23:
hello，这个 User_Setup.h  在你下载的库 TFT_eSPI-master  目录下，请将22行修饰掉，将52行的"//"去掉，抱歉，尽我所能
[image]

BojanJurca, 2020-03-23:
Thank you very much for your trouble. I believe it has something to do with Arduino since it doesn't correctly access even the first file #included <TFT_eSPI.h> - it takes another from somewhere else. I put the whole package into application folder and changed #include to "./TFT_eSPI-master/TFT_eSPI.h". I also had to change all nested "includes accordingly. 

At the end of TFT_eSPI.h I had to add #include "TFT_eSPI.cpp".

Now it works nicely. Thank you again.

closed 2020-03-23 by Bodmer