Bodmer/TFT_eSPI #580: Sprite flickering

Paulie92, 2020-03-23:
Hi Bodmer,

At first thank you for your library, great work!

I have a little problem with flickering while using a sprite. As the sprite shouldbe flicker-free it is quite strange for me. There is my code:

`spr2.fillSprite(TFT_BLACK);
 spr2.fillRect(position, 0, 8, 20, TFT_RED);
 spr2.pushSprite(0,190);
 delay(2);
`

At first look rendering on thedisplay looks OK, but I found out that the rectangle is sometimes not pushed on the display to the desired position in one step but it is "broken". It looks at one moment as  "S"). I took a 900fps video of the dislay: [https://youtu.be/ddwFlXiGTZI](url)

I am also using a mooving text rendered with sprite pushed on the dispaly and it is also sometimes broken:
[image]

Do you have any idea how this can be solved?  Thank you for any feedback.

Bodmer, 2020-03-23:
The screen is updated at the refresh rate by the TFT driver chip probably at 70-100 Hz (every 10-14 ms), so you are seeing the aliasing between the update rate in your sketch and the refresh rate of the screen.  Increase the delay between each Sprite draw to at least 25ms.

Paulie92, 2020-03-25:
As I have discovered more I am facing with the tearing effect. I tried to rise up the refresh cycle of my sketch to 30ms and the issue is still present. As I found out from the ST7789V datasheet, the panel refresh cycle is 14ms. 13 ms it is scanning its RAM and pushing to the panel and 1ms it is waiting for the new data:
[image]
[image]
So it does not depend how often you are uploading(drawing) the date to the dispay, because you will certainly hit the 13ms moment whe the panel is updating. As I understand, the only way how to prevent this is to synchronize uploading of the data to driver with the refreshing period of the panel and for this purpose has the driver "tearing effect" output line which is unfortunatelly not avaiable in my display. The another condition is, that the upload have to be faster then 14ms refresh cycle. Is the any other way how to avoid the tearing effect?

Another strange thing I found out is, that the pushing of 16-bit 240x240 sprite to the ST7789V display take 26ms (approx. twice the dispaly refresh rate, SPI @40Mhz), but when I run the same scatch on ILI9488 display, the pushing of the same sprite took 166ms! I assumed that the uploading time depends only on the speed of SPI which was 40Mhz in both cases). Is there any other aspect which influes the uploading time?

Bodmer, 2020-03-25:
You have understood the problem. You can mitigate the visual effect of the tearing by increasing the delay between the Sprite refreshes so that the image is shown as complete on the next display refresh cycle, persistance of vision then smooths out the tearing effect. This is why I suggested at least 25ms BETWEEN Sprite writes to the screen.

The ILI9488 requires 3 bytes per pixel rather than the usual 2 bytes, so performance suffers as a result. I have recently made some changes to improve performance for ILI9488 displays but the optimisation level is lower.

closed 2020-03-25 by Bodmer