Bodmer/TFT_eSPI #581: ILI9341 with STM32F103 Parallel 8-Bit Not working

HamidSaffari, 2020-03-25:
Hi, and salute to the creators.
I tried to setup ILI9341 Parallel 8-Bit with BluePill STM32 board but I realized that you only defined hard-coded pins for  NUCLEO_64_TFT and NUCLEO_144_TFT but defined not_hard-coded for others in line 599 of  TFT_eSPI_STM32.h :

////////////////////////////////////////////////////////////////////////////////////////
// Support for other STM32 boards (not optimised!)
////////////////////////////////////////////////////////////////////////////////////////
  #else
    // This will work with any STM32 to parallel TFT pin mapping but will be slower
........

but when you choose this condition compiler report errors as below:

In file included from C:\Users\Hamid\Documents\Arduino\libraries\TFT_eSPI\TFT_eSPI.cpp:24:
C:\Users\Hamid\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c: In member function 'void TFT_eSPI::busDir(uint32_t, uint8_t)':
C:\Users\Hamid\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:135:24: error: 'D0_PIN_PORT' was not declared in this scope
  135 |     LL_GPIO_SetPinMode(D0_PIN_PORT, D0_PIN_BIT, LL_GPIO_MODE_OUTPUT);
      |                        ^~~~~~~~~~~
C:\Users\Hamid\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:135:37: error: 'D0_PIN_BIT' was not declared in this scope
  135 |     LL_GPIO_SetPinMode(D0_PIN_PORT, D0_PIN_BIT, LL_GPIO_MODE_OUTPUT);
      |                                     ^~~~~~~~~~
C:\Users\Hamid\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:136:24: error: 'D1_PIN_PORT' was not declared in this scope
  136 |     LL_GPIO_SetPinMode(D1_PIN_PORT, D1_PIN_BIT, LL_GPIO_MODE_OUTPUT);
.
.
.
C:\Users\Hamid\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c: In member function 'uint8_t TFT_eSPI::readByte()':
C:\Users\Hamid\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:180:8: error: 'RD_TFT_D0' was not declared in this scope; did you mean 'TFT_D0'?
  180 |   b  = RD_TFT_D0 | RD_TFT_D0 | RD_TFT_D0 | RD_TFT_D0; //Delay for bits to settle
      |        ^~~~~~~~~
      |        TFT_D0
C:\Users\Hamid\Documents\Arduino\libraries\TFT_eSPI\Processors/TFT_eSPI_STM32.c:182:20: error: 'RD_TFT_D1' was not declared in this scope; did you mean 'TFT_D1'?
  182 |   b  = RD_TFT_D0 | RD_TFT_D1 | RD_TFT_D2 | RD_TFT_D3;
      |                    ^~~~~~~~~
      |                    TFT_D1
.
.
.
exit status 1
Error compiling for board Generic STM32F1 series.

 so I tried to resolve it like this and upload it to MCU but not worked:

////////////////////////////////////////////////////////////////////////////////////////
// Support for other STM32 boards (not optimised!)
////////////////////////////////////////////////////////////////////////////////////////
  #else
    // This will work with any STM32 to parallel TFT pin mapping but will be slower

    // Convert Arduino pin reference Dx or STM pin reference PXn to hardware port and mask
    #define D0_PIN_NAME  digitalPinToPinName(TFT_D0)
    #ifndef D0_PIN_PORT
      #define D0_PIN_PORT    get_GPIO_Port(STM_PORT(D0_PIN_NAME))
    #endif
    #define D0_PIN_MASK  (const uint16_t)STM_LL_GPIO_PIN(D0_PIN_NAME)<<16
    #define D0_PIN_BIT D0_PIN_MASK
	// Use bit set reset register
    #define D0_BSR(B) D0_PIN_PORT->BSRR = (D0_PIN_MASK)>>(B)

    // Convert Arduino pin reference Dx or STM pin reference PXn to hardware port and mask
    #define D1_PIN_NAME  digitalPinToPinName(TFT_D1)
    #ifndef D1_PIN_PORT
      #define D1_PIN_PORT    get_GPIO_Port(STM_PORT(D1_PIN_NAME))
    #endif
    #define D1_PIN_MASK  STM_LL_GPIO_PIN(D1_PIN_NAME)<<16
	#define D1_PIN_BIT D1_PIN_MASK
    // Use bit set reset register
    #define D1_BSR(B) D1_PIN_PORT->BSRR = D1_PIN_MASK>>(B)

    // Convert Arduino pin reference Dx or STM pin reference PXn to hardware port and mask
    #define D2_PIN_NAME  digitalPinToPinName(TFT_D2)
    #ifndef D2_PIN_PORT
      #define D2_PIN_PORT      get_GPIO_Port(STM_PORT(D2_PIN_NAME))
    #endif
    #define D2_PIN_MASK  (const uint16_t)STM_LL_GPIO_PIN(D2_PIN_NAME)<<16
    #define D2_PIN_BIT D2_PIN_MASK
	// Use bit set reset register
    #define D2_BSR(B) D2_PIN_PORT->BSRR = D2_PIN_MASK>>(B)

    // Convert Arduino pin reference Dx or STM pin reference PXn to hardware port and mask
    #define D3_PIN_NAME  digitalPinToPinName(TFT_D3)
    #ifndef D3_PIN_PORT
      #define D3_PIN_PORT      get_GPIO_Port(STM_PORT(D3_PIN_NAME))
    #endif
    #define D3_PIN_MASK  (const uint16_t)STM_LL_GPIO_PIN(D3_PIN_NAME)<<16
    #define D3_PIN_BIT D3_PIN_MASK
	// Use bit set reset register
    #define D3_BSR(B) D3_PIN_PORT->BSRR = D3_PIN_MASK>>(B)

    // Convert Arduino pin reference Dx or STM pin reference PXn to hardware port and mask
    #define D4_PIN_NAME  digitalPinToPinName(TFT_D4)
    #ifndef D4_PIN_PORT
      #define D4_PIN_PORT      get_GPIO_Port(STM_PORT(D4_PIN_NAME))
    #endif
    #define D4_PIN_MASK  (const uint16_t)STM_LL_GPIO_PIN(D4_PIN_NAME)<<16
    #define D4_PIN_BIT D4_PIN_MASK
	// Use bit set reset register
    #define D4_BSR(B) D4_PIN_PORT->BSRR = D4_PIN_MASK>>(B)

    // Convert Arduino pin reference Dx or STM pin reference PXn to hardware port and mask
    #define D5_PIN_NAME  digitalPinToPinName(TFT_D5)
    #ifndef D5_PIN_PORT
      #define D5_PIN_PORT      get_GPIO_Port(STM_PORT(D5_PIN_NAME))
    #endif
    #define D5_PIN_MASK  (const uint16_t)STM_LL_GPIO_PIN(D5_PIN_NAME)<<16
    #define D5_PIN_BIT D5_PIN_MASK
	// Use bit set reset register
    #define D5_BSR(B) D5_PIN_PORT->BSRR = D5_PIN_MASK>>(B)

    // Convert Arduino pin reference Dx or STM pin reference PXn to hardware port and mask
    #define D6_PIN_NAME  digitalPinToPinName(TFT_D6)
    #ifndef D6_PIN_PORT
      #define D6_PIN_PORT      get_GPIO_Port(STM_PORT(D6_PIN_NAME))
    #endif
    #define D6_PIN_MASK  (const uint16_t)STM_LL_GPIO_PIN(D6_PIN_NAME)<<16
    #define D6_PIN_BIT D6_PIN_MASK
	// Use bit set reset register
    #define D6_BSR(B) D6_PIN_PORT->BSRR = D6_PIN_MASK>>(B)

    // Convert Arduino pin reference Dx or STM pin reference PXn to hardware port and mask
    #define D7_PIN_NAME  digitalPinToPinName(TFT_D7)
    #ifndef D7_PIN_PORT
      #define D7_PIN_PORT      get_GPIO_Port(STM_PORT(D7_PIN_NAME))
    #endif
    #define D7_PIN_MASK  (const uint16_t)STM_LL_GPIO_PIN(D7_PIN_NAME)<<16
    #define D7_PIN_BIT D7_PIN_MASK
	// Use bit set reset register
    #define D7_BSR(B) D7_PIN_PORT->BSRR = D7_PIN_MASK>>(B)

    // Write 8 bits to TFT for any Nucleo
    #define tft_Write_8(C) D7_BSR(((C)>>3)&0x10); D6_BSR(((C)>>2)&0x10); \
                           D5_BSR(((C)>>1)&0x10); D4_BSR(((C)>>0)&0x10); \
                           WR_L; \
                           D3_BSR(((C)<<1)&0x10); D2_BSR(((C)<<2)&0x10); \
                           D1_BSR(((C)<<3)&0x10); D0_BSR(((C)<<4)&0x10); \
                           WR_H

    // Write 16 bits to TFT
    #define tft_Write_16(C) tft_Write_8((C)>>8); tft_Write_8(C)

    // 16 bit write with swapped bytes
    #define tft_Write_16S(C) tft_Write_8(C); tft_Write_8((C)>>8)

    // Read a data bit
    #define RD_TFT_D0 (((D0_PIN_PORT->IDR)&(D0_PIN_MASK))>>(D0_PIN_MASK-0))
    #define RD_TFT_D1 (((D1_PIN_PORT->IDR)&(D1_PIN_MASK))>>(D1_PIN_MASK-1)) 
    #define RD_TFT_D2 (((D2_PIN_PORT->IDR)&(D2_PIN_MASK))>>(D2_PIN_MASK-2)) 
    #define RD_TFT_D3 (((D3_PIN_PORT->IDR)&(D3_PIN_MASK))>>(D3_PIN_MASK-3)) 
    #define RD_TFT_D4 (((D4_PIN_PORT->IDR)&(D4_PIN_MASK))>>(D4_PIN_MASK-4)) 
    #define RD_TFT_D5 (((D5_PIN_PORT->IDR)&(D5_PIN_MASK))>>(D5_PIN_MASK-5)) 
    #define RD_TFT_D6 (((D6_PIN_PORT->IDR)&(D6_PIN_MASK))>>(D6_PIN_MASK-6)) 
    #define RD_TFT_D7 (((D7_PIN_PORT->IDR)&(D7_PIN_MASK))>>(D7_PIN_MASK-7)) 
	
  #endif // End of parallel/Nucleo 64/144 selections
//----------------------------------------------------------------------------

what I have doubted is: does PIN_BIT equal to PIN_MASK ?

and I'm sure about my hardware and connections because I've got the result by this library:
https://github.com/stevstrong/Adafruit_TFTLCD_8bit_STM32   although he used Roger Clark core.

Bodmer, 2020-03-25:
I have updated the library so this should now be fixed. Please report back if you have problems and tell me the pin allocation you have used.

HamidSaffari, 2020-03-26:
Thanks man, but still show nothing in screen.
I checked with oscilloscope and there is no transaction or pulses in TFT_DC and TFT_WR pins but there were some slow repeatable pulses in other pins.

My connections:
#define TFT_CS   PB7  // Chip select control pin
#define TFT_DC   PB8  // Data Command control pin
#define TFT_RST  PB0  // Reset pin
#define TFT_WR   PB9  // Write strobe control pin 
#define TFT_RD   PB1  // Read pin
#define TFT_D0   PA0  // 8 bit parallel bus to TFT
#define TFT_D1   PA1
#define TFT_D2   PA2
#define TFT_D3   PA3
#define TFT_D4   PA4
#define TFT_D5   PA5
#define TFT_D6   PA6
#define TFT_D7   PA7

and it works with https://github.com/stevstrong/Adafruit_TFTLCD_8bit_STM32

closed 2020-03-27 by Bodmer

Bodmer, 2020-03-27:
I have updated the library so it should work now. I have added a new configuration option to take advantage of the direct port A mapping to the data bus, this makes it run faster.

This is the setup I used:

```
// Define STM32 to invoke STM32 optimised driver
#define STM32

// Define if Port A pins 7-0 are used for data bus bits 7-0
#define STM_PORTA_DATA_BUS

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define TFT_PARALLEL_8_BIT

// Define the TFT display driver
#define ILI9341_DRIVER

#define TFT_CS PB7 // Chip select control pin
#define TFT_DC PB8 // Data Command control pin
#define TFT_RST PB0 // Reset pin
#define TFT_WR PB9 // Write strobe control pin
#define TFT_RD PB1 // Read pin
#define TFT_D0 PA0 // 8 bit parallel bus to TFT
#define TFT_D1 PA1
#define TFT_D2 PA2
#define TFT_D3 PA3
#define TFT_D4 PA4
#define TFT_D5 PA5
#define TFT_D6 PA6
#define TFT_D7 PA7

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// STM32 support for smooth fonts via program memory (FLASH) arrays
#define SMOOTH_FONT

```

HamidSaffari, 2020-03-27:
Still I've got nothing on screen. even with ILI9488 LCD in parallel connection.

HamidSaffari, 2020-03-27:
When I run your TFT_Char_times.ino code in both SPI and 8-bit-Parallel mode on blue pill, the result is 5 times faster in parallel mode.
SPI:                   Biggest font:  5.3 ms
8-bit-Parallel:   Biggest font:  1.4 ms
so in High resolution and fast refreshing projects like LVGL definitely parallel is better solution.
Now can you consider implementing a 16-bit-Parallel mode so we can go faster? like this library:
https://github.com/stevstrong/Adafruit_TFTLCD_16bit_STM32

HamidSaffari, 2020-03-27:
And also without using STM_PORTA_DATA_BUS its getting so slow:
8-bit-Parallel without using STM_PORTA_DATA_BUS : Biggest font draw time: 7.7 ms

Bodmer, 2020-03-27:
I have no plans to add 16 bit compatibility or FSMC support.
If random pins are asigned to the data bits a lot of shift+And +8 GPIO writes is needed so using port A or B bits 0-7 is best to reduce the per write overhead. See 8 bit write to port [here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.h#L633) compared to individual bits https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.h#L633.

Bodmer, 2020-03-27:
You mention LittleVGL, the example I saw for the ESP32 uses a very inefficient screen update method. Something like this would be better:

```
void my_disp_flush(lv_disp_drv_t *disp, const lv_area_t *area, lv_color_t *color_p)
{
  uint32_t w = (area->x2 - area->x1 + 1);
  uint32_t h = (area->y2 - area->y1 + 1);

  tft.startWrite();
#ifdef USE_DMA
  tft.pushImageDMA(area->x1, area->y1, w, h, &color_p->full);
  // Note: tft.endWrite(); must be called in my_lv_tick_handler when DMA is complete
#else
  tft.setAddrWindow(area->x1, area->y1, w, h);
  tft.pushColors(&color_p->full, w * h, true);
  tft.endWrite();
#endif

  lv_disp_flush_ready(disp);
}
```

Pablo2048, 2020-03-27:
Hi Bodmer,
thanks for LVGL `my_disp_flush()` enhancement - I'll give it a try and modify the example(s)...

commit 89bf0ce6 (2020-03-25): Fix #581 plus

Update 4 bit Sprite code
Update Animate_Dial example to use italic font with minimised character set, add original jpeg graphic.
Raise to 2.1.8
commit d1b0bab9 (2020-03-25): Fix #581
commit 919febef (2020-03-27): Fix #581