Bodmer/TFT_eSPI #583: ILI9488 8-Bit-Parallel interface with STM32F103 Not working

HamidSaffari, 2020-03-27:
Hi, Bodmer.
I tried to setup 3.95" 480x320  ILI9488 8-Bit-Parallel nterface with BluePill STM32 board but have got nothing but this:

[image]

https://www.aliexpress.com/item/32845972464.html
[image]

And my TFT setup:

// Define STM32 to invoke STM32 optimised driver
#define STM32

// Define if Port A pins 7-0 are used for data bus bits 7-0
#define STM_PORTA_DATA_BUS

// Tell the library to use 8 bit parallel mode (otherwise SPI is assumed)
#define TFT_PARALLEL_8_BIT

// Define the TFT display driver
//#define ILI9341_DRIVER
#define ILI9488_DRIVER

#define TFT_CS PB7 // Chip select control pin
#define TFT_DC PB8 // Data Command control pin
#define TFT_RST PB0 // Reset pin
#define TFT_WR PB9 // Write strobe control pin
#define TFT_RD PB1 // Read pin
#define TFT_D0 PA0 // 8 bit parallel bus to TFT
#define TFT_D1 PA1
#define TFT_D2 PA2
#define TFT_D3 PA3
#define TFT_D4 PA4
#define TFT_D5 PA5
#define TFT_D6 PA6
#define TFT_D7 PA7

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:-.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

// STM32 support for smooth fonts via program memory (FLASH) arrays
#define SMOOTH_FONT

Bodmer, 2020-03-27:
I have updated the init code for the ILI9488 in the repository, try it again.

The write strobe timing for the ILI9488 has not been optimised yet so it will be slower than the ILI9431. I will look at that later today.

HamidSaffari, 2020-03-27:
Nope, same result.

Bodmer, 2020-03-27:
Check you really have an ILI9488
Check you are using the latest copy of the library which I updated recently.

The ILI9488 works fine on my setup so I am unable to reproduce the problem.

Bodmer, 2020-03-27:
Make sure you copy of the library [has this change](https://github.com/Bodmer/TFT_eSPI/blob/master/TFT_Drivers/ILI9488_Init.h#L61).

closed 2020-03-27 by Bodmer

HamidSaffari, 2020-03-29:
But This LCD works with https://github.com/prenticedavid/MCUFRIEND_kbv  on Arduino UNO:

[image]
[image]
[image]

HamidSaffari, 2020-03-29:
By the way in ILI9488   WRX twc spec is 40ns = 25MHz  if you want to add it's TWRL.

Bodmer, 2020-03-29:
Here is my test setup:
[image]

[image]

I am using [setup file 35](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.c#L131-L146) with the ILI9488 driver enabled.

It is running fine. Check your 5V power is sufficient, power via the USB port.

MCUFRIEND_kbv also works on Blue Pill boards with the ILI9488 so you could try that to verify your connections are OK.