Bodmer/TFT_eSPI #588: Button overdraw in Keypad example

mmmilan, 2020-03-28:
Hi!

This is my first touch display, so I might be missing something important. Could someone help me, please?

This is an ILI9341 connected to ESP32. I'm using Arduino IDE 1.8.12, ESP32 1.0.4, TFT_eSPI 2.1.4. I've uploaded Keypad_240x320.ino, calibration seems OK, and the keypad shows up properly:
[image]

But when I press 1:
[image]

And then 5:
[image]

And then 9:
[image]
...the buttons draw over each other and everything.

videobuff, 2020-03-29:
I have exactly the same issue. Latest library from github

Verstuurd vanaf mijn telefoon
Groeten
Erik

Bodmer, 2020-03-29:
Thanks for reporting this.
Should now be fixed.

closed 2020-03-29 by Bodmer

mmmilan, 2020-03-30:
It's fixed. Thank you!

commit 5ab0a08d (2020-03-29): Fix #588