Bodmer/TFT_eSPI #589: STM32 GPIO TypeDef problem

HamidSaffari, 2020-03-29:
Your new change causes this problem on STM32:
.......    TFT_eSPI\Processors/TFT_eSPI_STM32.c:132:30: error: 'struct GPIO_TypeDef' has no member named 'MODER'; did you mean 'ODR'?
  132 |   if (mode == OUTPUT) GPIOA->MODER = (GPIOA->MODER & 0xFFFF0000) | 0x00005555;

Which I think MODER must change to ODR.

HamidSaffari, 2020-03-29:
Also consider defining PORTC Data bus fast DIR like A and B as well. because Both PORTA and PORTB Bit0-7 have some important 2nd functionalities. eg: PA4-PA7: SPI1. PB3,PB4: JTAG.
Of course PORTC Bit0-7 not available in Bluepill (32F103C), but on other STM boards.

Bodmer, 2020-03-29:
Thanks for reporting this. The compile error has been fixed now. [See here](https://github.com/Bodmer/TFT_eSPI/blob/master/Processors/TFT_eSPI_STM32.c#L131-L146). It appears that the STM32F1xx series is unique in that it does not support MODER.

I understand that Port B 0-7 has useful functionality and all these pins are not all available on the "Blue Pill". However I prefer the STM32F411CE "Black Pill" variant which has more RAM and FLASH, it does break out the Port B0-7 and this allows me to free up the analogue channels on Port A. Hence the inclusion of this feature.

It is best to use SPI displays or a higher pin count board if multiple port capability is needed on low pin count boards as the parallel display interface burns up so many pins. SPI also has the benefit that DMA is supported to boost performance.

closed 2020-03-29 by Bodmer