Bodmer/TFT_eSPI #591: Euro Symbol not working with Anti Aliased Fonts

Redstoned, 2020-04-06:
Hi,
I created an Anti-Aliased font with the processing tool.
I included currency symbols in the .vlw file: (unicodeBlocks)
` 0x20A0, 0x20CF, //Currency Symbols, 48, 32, Common`

The font loads fine onto my ESP32 and it also prints normal ASCII characters and special chars as 'ä', 'ö', 'ü', etc. I also included,
but sadly, if I want to draw a "€"-Symbol, it draws the rectangle-not  found Symbol.

Am I missing something that I cannot draw this Symbol?

Bodmer, 2020-04-06:
The Euro symbol is 0x20AC, so these should work:

tft.println("€");      // €
tft.println("\u20AC"); // €

Redstoned, 2020-04-06:
I used the drawString-function previously and that didn't work, but the println function does work for me. That's pretty strange, isn't it?

Bodmer, 2020-04-06:
Check the right symbol was being used, note that unicode 0x20AC was not listed in your original post as being loaded in the font.

Post an example sketch that is not working for you complete with font file.

Bodmer, 2020-04-06:
I have run your sketch above on an ESP32 with your font file and it ran fine. I guess we have a different environment, I am runing:
ESP32 board package: 1.0.4
Arduino IDE: 1.8.11
TFT_eSPI version: 2.2.0

What is your setup?

Redstoned, 2020-04-07:
I realised I used version 1.3.3 of your library. I updated to the newest version and now it works just fine. Now I looked at the releases and I think the bug got fixed in version 1.4.8.
I am sorry for my mistake.

closed 2020-04-07 by Redstoned

Bodmer, 2020-04-07:
Glad to hear you have it working.