Bodmer/TFT_eSPI #593: Support Pauls fonts?

tablatronix, 2020-04-09:
Would it be possible to support PaulStoffregens fonts?
https://github.com/PaulStoffregen/ILI9341_fonts

I would not think there would be much work required if any, not sure what format they are in..

Render code
https://github.com/PaulStoffregen/ILI9341_t3/blob/master/ILI9341_t3.cpp#L1476

Bodmer, 2020-04-09:
Thanks for the suggestion.

I have looked at Paul's fonts in the past and decided the format was too restrictive. TFT_eSPI already supports 4 font formats the GFX compatible fonts can be generated by a PC (see compiled command line code[ here](https://github.com/Bodmer/GFX_Font_Converter) for Windows) or there are also online tools to generate fonts.  I tend to use anti-aliased (smooth) fonts now as they look so much better but they are greedy on FLASH, but processors with large FLASH resources are low cost these days. Smooth fonts can be created by using the "Processing" sketch in the library tools folder (https://github.com/Bodmer/TFT_eSPI/tree/master/Tools/Create_Smooth_Font) from your own computers fonts. Smooth fonts allow any mix of characters from the 16 bit Unicode range so it is easy to just include large numbers for example or any individual mix of characters.

TFT_eSPI also supports the 16 bit Unicode range for Adafruit GFX fonts but this has the same restriction as Paul's font format in that you can only load blocks not a random selection of characters.

The only font format I might consider adding at a future date is that of LittlevGL as that format is flexible and not restrictive.

tablatronix, 2020-04-09:
I noticed smooth fonts flicker alot, not sure how to alleviate it or what causes it, might be tearing.

I guess I will generate some glyph fonts then, I noticed the free fonts are lower ascii only, so I will need a separate font for symbols. Or just use the gfx ones. Want to see if I can get a smooth font awesome font working, should be fine if they are not updating often, we will see.

Bodmer, 2020-04-09:
The smooth font examples "Font_Demo_x_Array" where x is 1-4 show different methods of using smooth fonts including the use of sprites to reduce flicker. Tearing can be an issue for very fast changes such as a fast incrementing number but to be readable by humans numbers should not be changed rapidly and not updated if the number has not changed.

Check out #290 as the font awesome has been used successfully.

closed 2020-04-14 by Bodmer

tablatronix, 2020-04-17:
Thanks , I will test out the sprites, I did figure out most of my flicker problem..

I was leaving textPadding what I had used it as last and not reseting it, so some drawStrings were reedrawing all over the place.. It would be nice to be able to pass padding into the draw methods as an argument to prevent things like this, its harder to forget change it when you can pass it as a param..