Bodmer/TFT_eSPI #597: Arduino 32 bit Support

el-samiyel, 2020-04-14:
Hi,

Does this lib support any of the arduino atmel 32 but chips?

Many Thanks

Bodmer, 2020-04-14:
Yes, but there is no low level driver optimisation for Atmel processors so it will run more slowly.

Which processor board are you thinking of using?

el-samiyel, 2020-04-14:
> Yes, but there is no low level driver optimisation for Atmel processors so it will run more slowly.
Excellent, understood. I am still awaiting delivery of my screen, so have just been studying your code ready for testing.
> Which processor board are you thinking of using?
I preferably wanted to use the Zero, with the 7789 display but if not the due will do.

Do you have any tips on drawing jpgs from an SD card, are there any special requirements that should be configured in the User_Setup.h such as the SPI freq?

Thanks again.

Bodmer, 2020-04-15:
There are no special requirements for rendering jpgs other than having sufficient RAM and most 32 bit processors have plenty.
The maximum SPI frequency may be constrained by the processor you use. Typically frequencies up to 40MHz work for writes to the TFT screen. If you get occasional bad pixels reduce the frequency to 27MHz.

closed 2020-04-15 by Bodmer

el-samiyel, 2020-07-05:
Hi, I have finally received my screen but am having no luck loading images from the sd card on a arduino zero.

Do you have any example on how to do this? As when I load the ESP32_SDcard_jpeg i get 'FS.h No such file or directory'.

Cheers

Bodmer, 2020-07-05:
First get the screen working with simple library examples and with no SD card connected. Next get the SD card working with the TFT CS line connected to 3.3V using the SD card library examples only. Next try an SD Card example with the TFT library loaded. Make sure you set both SD and TFT chip selects HIGH at the start of setup() before you use sd or tft begin functions because a floating CS line can prevent Sd or tft initialisation.

el-samiyel, 2020-07-05:
Thanks for the quick reply.

So I have the screen working, and I'm able to get the examples working. I also, have the sd card and screen working at the same time. My issue is that i get 'FS.h NO such file or directory' when loading the ESP32_SDcard_jpeg sketch. Is the FS.h lib for the ESP32 only?

Thanks

Bodmer, 2020-07-05:
OK, understood.

Yes FS.h is ESP32 specific, hence the naming of the sketch. You can probably just delete that line but you will have to adapt/comment out other lines such as the card type functions which may also be unique to the ESP32.