Bodmer/TFT_eSPI #598: [Feature] Smooth font as a shared class

adamczuk-piotr, 2020-04-14:
Hi, 

Is there any obstacle that that  the smooth  font cannot be an instance of separate class and shared between TFT and Sprites?
I know that it consumes a lot of RAM but on esp8266 i can handle up to four founts loaded at a time and have some space to drawe Sprite of 240px wide with them up to 48px in height. 

The max I can get now is 3 fonts loaded at a time. 
Using some kind of workaround that I have three diifrent Spites with loaded fonts and drawing on Sprite with 240 width and 72 px height.
The limit is around 20k of free contiunus RAM space (so 1/4 of all) to create Sprite with 8bit colorspace.

The number of "Fonts" loaded can be configured.

I will try it by my own to move move this smooth font to separate class and that it can ve shared with TFT and Sprite.

But just to know, if somebody try do that and why not?

Bodmer, 2020-04-15:
I did consider this but it is simpler to build the font handling into the TFT_eSPI base class. You will find changing it is more intrusive for code changes than you might expect. Note that the Sprite class has it's own built in smooth font glyph rendering for performance reasons.

closed 2020-04-15 by Bodmer

adamczuk-piotr, 2020-04-15:
I've managed to make it working, at least from SPIFFS in esp8266. Thanks for your work!
Now I have to think about font management and some unicode substrings for my scrolling labels.