Bodmer/TFT_eSPI #599: Support of 16 bit parallel bus for ESP32

Jakub-Nagy, 2020-04-15:
Hi guys,
I'm building a device with a 3.5" 480x320 CTP touchscreen with ILI9488 driver. This is its datasheet:
[JLT35002A-PSS.pdf](https://github.com/Bodmer/TFT_eSPI/files/4480590/JLT35002A-PSS.pdf)

This is the schematic:

[image]

So far everything is tested and works great. The problem is the long update periods when scrolling on the screen.  So I want to switch from SPI to 8 bit parallel or preferably even to 16 bit parallel. Does this library support 16 bit parallel interface for the ESP32? Because I cannot find the setup for this in the User_Setup.h. If so, how can I initialize it correctly? Are there any hardware requirements for connecting the 16 bit bus or can I just use any of the GPIOs?

Lastly, how much faster would be the display update period when using 16 bit instead of 8 bit? A straightforward answer might be double but I'm not sure if that's the case. Also, does the ESP32 core catch up to the high speed bus or would I need to switch the core oscillator to one with a higher frequency?

Thank you very much.

Bodmer, 2020-04-15:
The ILI9488 in SPI mode requires 3 bytes per pixel compared tot he usual 2 bytes for most displays so this has a performance impact. In 8 bit parallel mode only 2 bytes per pixel is needed.

In portrait orientation the displays have a hardware scroll facility but this must be managed by the sketch, see TFT_Terminal example and look at the display data sheet.

I will not be adding 16 bit parallel support to the library.

Here are the results for SPI at 27MHz with Setup21...
```
Benchmark,                Time (microseconds)
Screen fill (5 times),    731530
Text,                     26404
Lines,                    294855
Horiz/Vert Lines,         60919
Rectangles (outline),     33851
Rectangles (filled),      1783469
Circles (filled),         190270
Circles (outline),        190130
Triangles (outline),      59212
Triangles (filled),       586454
Rounded rects (outline),  92615
Rounded rects (filled),   1785505
Total = 5835214us
Total = 5.8352s
```

Here are the performance results for the 8 bit parallel Setup16...
```
Benchmark,                Time (microseconds)
Screen fill (5 times),    201365
Text,                     6534
Lines,                    78215
Horiz/Vert Lines,         18960
Rectangles (outline),     10578
Rectangles (filled),      377643
Circles (filled),         57280
Circles (outline),        39964
Triangles (outline),      17382
Triangles (filled),       184517
Rounded rects (outline),  23319
Rounded rects (filled),   566291
Total = 1582048us
Total = 1.5820s
```
No changes to the ESP32 processor frequency are required.

closed 2020-04-15 by Bodmer

Jakub-Nagy, 2020-04-15:
Thank you for the prompt reply. I will use the 8 bit bus then.

I have one more question. Are the screen fills 5 continuous ones? Can I divide the time by 5 to get an average? So in 8 bit, should I expect time for one screen fill to be 201365 / 5 = 40273 us?

Bodmer, 2020-04-15:
Yes, your calculations are correct, it takes approx. 40ms to clear the screen.

jamon, 2020-09-15:
@Bodmer 
Apologies for resurrecting a dead thread, but I felt it better than creating a new one in this case.

Regarding 16bit parallel support, I'm not an expert in C/C++, but glancing over the code, it looks as though you're calling 16bit write functions that delegate to the 8bit parallel writes or SPI writes as needed for the driver.  That gives me the impression that adding 16bit parallel support for a particular driver/mcu would be relatively straightforward.

Based on your response above, though, I feel like I'm missing something that would lead you to not wanting to add support for 16bit interfaces.  I have a 16bit display (ILI9486-based) and a variety of boards (stm32f466, stm32f103, esp32, arduinos, pis, etc).  If it's not too much trouble, would you mind pointing me at what the major issues I'll likely run into are?

Thanks!

vijaysam13, 2021-05-24:
Hii, i recently bought a 3.2 tft lcd display with ili9341 driver with 16 bit interface (no spi interface), could you please guide me to use this display with esp 32
Pin out of display below:
[image]