Bodmer/TFT_eSPI #602: WaveShare 3.2" ESP32-WROOM-32, cannot make it work

dchaplinsky, 2020-04-19:
Hello.

Sorry in advance for the noob question. I have one of 3.2 RPi Waveshare clones and ESP32 (38 pin version).

I've uncommented `User_Setups/Setup36_RPi_touch_ILI9341.h` in User_Setup_Select.h and connected it according to https://github.com/Bodmer/TFT_eSPI/blob/master/Tools/RPi_TFT_Connections.png

White screen is all I've got so far. Maybe, I've selected wrong chipset? (here is display page: https://arduino.ua/prod1768-tft-lcd-sensornii-displei-3-2-na-ili9341-dlya-rasbberry-pi)

What I'm doing wrong? I'm already slightly depressed and want to throw both boards out of window :)
[image]
[image]

dchaplinsky, 2020-04-19:
[code]
TFT_eSPI ver = 2.2.0
Processor    = ESP50
Frequency    = 240 MHz
Transactions = Yes 
Interface    = SPI 
Display driver = 9341
Display width  = 240 
Display height = 320 

MOSI    = D-1 (GPIO 23)
MISO    = D-1 (GPIO 19)
SCK     = D-1 (GPIO 18)
TFT_CS   = D8 (GPIO 15)
TFT_DC   = D4 (GPIO 2)
TFT_RST  = D2 (GPIO 4)

TOUCH_CS = D-1 (GPIO 22)

Font GLCD   loaded
Font 2      loaded
Font 4      loaded
Font 6      loaded
Font 7      loaded
Font 8      loaded
Smooth font enabled

Display SPI frequency = 40.0 MHz 
Touch SPI frequency   = 2.5 MHz 
[/code]

Bodmer, 2020-04-19:
It appears it has not picked up your setup.

You must [comment out this line...](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L22)

I would try [setup 11](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L35).

You may not be able to get that display to work as it does not that the serial to parallel converter chips.

You might get this setup to work (use it exactly as written and do NOT specify the RPi display type as that is for a display with the serial to parallel chips fitted):
```
#define ILI9431_DRIVER

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS   15  // Chip select control pin
#define TFT_DC    2  // Data Command control pin
#define TFT_RST   4  // Reset pin (could connect to RST pin)
//#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 22     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT

#define SPI_FREQUENCY  40000000

#define SPI_TOUCH_FREQUENCY  2500000
```

thush, 2020-04-19:
Thanks for Quick revert back.. I am also looking into this..will try

Bodmer, 2020-04-19:
Setup 11 will not work as it appears you have a 240x320 display and not one fitted with ILI9486.

dchaplinsky, 2020-04-19:
Thanks for quick reply, @Bodmer.

I'll try example above shortly. Also not sure about ILI9486. Display description on webpage above says 9341, but I don't know how accurate it is. Seems that I have a clone of Waveshare RPi, but waveshare's page has no info about chip as well

thush, 2020-04-19:
Hi,
 I have one of 2in9 Waveshare  (8 pin version)-296×128 pixels display.
I have included   "epd2in9.h" and needed .h .  The TestCard.bmp refered on the released doesn't work.

Bodmer, 2020-04-19:
@thush: Open a new issue. This topic is RPi TFT displays.

dchaplinsky, 2020-05-01:
Hello @Bodmer.
I've finally found a time to play with it. Typo in your config puzzled me a bit (9431 rather than 9341 :))

Anyway, it didn't work for me. Will try to play more with that device, but so far :((

dchaplinsky, 2020-05-01:
My output for Read_User_Setup, just in case:
```
17:24:13.179 -> 
17:24:13.179 -> [code]
17:24:13.179 -> TFT_eSPI ver = 2.2.0
17:24:13.179 -> Processor    = ESP50
17:24:13.179 -> Frequency    = 240 MHz
17:24:13.179 -> Transactions = Yes 
17:24:13.179 -> Interface    = SPI 
17:24:13.179 -> Display driver = 9341
17:24:13.179 -> Display width  = 240 
17:24:13.179 -> Display height = 320 
17:24:13.179 -> 
17:24:13.179 -> MOSI    = D-1 (GPIO 23)
17:24:13.179 -> MISO    = D-1 (GPIO 19)
17:24:13.179 -> SCK     = D-1 (GPIO 18)
17:24:13.179 -> TFT_CS   = D8 (GPIO 15)
17:24:13.212 -> TFT_DC   = D4 (GPIO 2)
17:24:13.212 -> TFT_RST  = D2 (GPIO 4)
17:24:13.212 -> 
17:24:13.212 -> TOUCH_CS = D-1 (GPIO 22)
17:24:13.212 -> 
17:24:13.212 -> Font GLCD   loaded
17:24:13.212 -> Font 2      loaded
17:24:13.212 -> Font 4      loaded
17:24:13.212 -> Font 6      loaded
17:24:13.212 -> Font 7      loaded
17:24:13.212 -> Font 8      loaded
17:24:13.212 -> Smooth font enabled
17:24:13.212 -> 
17:24:13.212 -> Display SPI frequency = 40.0 MHz 
17:24:13.212 -> Touch SPI frequency   = 2.5 MHz 
17:24:13.212 -> [/code]
```

Bodmer, 2020-06-15:
Unfortunately the RPI displays come in many variants and little documentation other than setting up for a RPi. It looks like yours is not compatible with the library.

closed 2020-06-15 by Bodmer