Bodmer/TFT_eSPI #604: ePaper2in9 display Run issue

thush, 2020-04-19:
I have included the needed epd2in9 headers for the Display type.
This line was uncommented as below in User_setup_select.h ;
#include <../../TFT_eSPI/User_Setups/Setup17_ePaper.h>            // Setup file for any Waveshare ePaper display

The IDE complains always now;

C:\Users\rqvn43\Documents\Arduino\libraries\TFT_eSPI_Setups/User_Setup_Select.h:116:15: fatal error: TFT_Drivers/ILI9341_Defines.h: No such file or directory
      #include <TFT_Drivers/ILI9341_Defines.h>

What i am doing wrong in running ?

thush, 2020-04-19:
Full Error to help me out:
----------------------------------------------------------------------
In file included from sketch\TFT_eSPI.h:35:0,
from C:\Users\rqvn43\Documents\Arduino\TFT_eSPI\examples\ePaper\Floyd_Steinberg\Floyd_Steinberg.ino:43:
C:\Users\rqvn43\Documents\Arduino\libraries\TFT_eSPI_Setups/User_Setup_Select.h:116:15: fatal error: TFT_Drivers/ILI9341_Defines.h: No such file or directory
      #include <TFT_Drivers/ILI9341_Defines.h>
               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

compilation terminated.

Bodmer, 2020-04-20:
[See here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup_Select.h#L20-L22). Only one setup should be included so comment out line 22

thush, 2020-04-20:
The setup_file porting headers  to /library folder is essential ?
Isn't there anyway around that ? Just to run the example as it is from the cloned repo ?

thush, 2020-04-20:
Really like your architecture setup for  Bit manipulation and e Paper support , but I am still unable to run the same with the full Setup above..
Is there anyway only BMP example can be run, without the dependencies and bulk of the libs ?

Thanks in advance.

Bodmer, 2020-04-20:
The #defined in the setup file eliminate a lot of options in the sketch and make the sketch more compact. Also once the library is setup for a particular configuration, then in the case of the TFT all relvant examples run without modification.

There are options that reduce the code loaded (fonts for example) but typically program memory limits are not an issue. The compiler will not include functions you do not use.

The ESP library example is the simplest and does not use TFT_eSPI at all, but then you do not get many features.

Do you have the display working at all?

thush, 2020-04-21:
I tried with my team mate, and we cant get the code compiled to any ePaper
example.
After commenting out #Include <Fonts> errors in the TFT, errors on  <ES.h>
will not go away.
We gave up afterwards :( '
Can you point to a simpler method ?

Bodmer, 2020-04-21:
There apperas to be a configuration problem on your PC so we need to go through this step by step. To start:
1. What version of the Arduino IDE arr you using?
2. What processor are you using to interface with the EPD?

thush, 2020-04-25:
-Arduino 1.8.12
-I have a Arduino-UNO and its plugged to Epaper 2in9(waveshare)

:)

Bodmer, 2020-04-25:
Unfortunately the UNO does not have enough RAM to use the TFT_eSPI library with an ePaper display as it uses a full screen buffer.  I would look for another library that uses paged memory. Maybe[ this one](https://github.com/ZinggJM/GxEPD2).

Be aware that most ePaper displays are only compatible with 3.3V logic levels and the UNO uses 5V levels which may damage the display.

closed 2020-04-25 by Bodmer