Bodmer/TFT_eSPI #605: Whitescreen on RPI 3.5" display with ESP32

dimosap, 2020-04-20:
Hello, first of all thank you for the great library. I've been trying to use a RPI 3.5" display with my esp32. I tried every available setup that matched my hardware but most of the time all i got was a white screen (rarely with some glitches). Sometimes i managed to view the 3D cube demo, then on another example my results were unsuccessful (funny thing though, after getting yet another white screen i tried to reupload the cube sketch but it didn't work). I also thought it could be a power supply issue but upon connecting my display on an external  power source my problem remained. Finally my pin connections follow the setup on Setup11_RPi_touch_ILI9486.h and my display can be seen here: github.com/dimosap/rpi-display/blob/master/display.jpg
I hope it's something stupid i'm missing, thank you in advance!

Bodmer, 2020-04-20:
Try reducing the SPI clock rate down to 16MHz in the setup file:

```
#define SPI_FREQUENCY  16000000
```

dimosap, 2020-04-20:
Thanks for your quick response. On SPI_FREQUENCY to 16000000 my display is still glitching :(

Bodmer, 2020-04-20:
Post a link to the vendors web page and I will have a look and see if it is likely to be compatible.

dimosap, 2020-04-21:
It's an old one i bought from banggood and even after a long search i failed to find the vendor. But since it did work sometimes maybe i'm missing something on the header files and it's not a compatibility issue? 

 I commented the original User_Setup.h from the User_Setup_Select.h and selected the Setup11_RPi_touch_ILI9486.h. It's content can be seen below (just in case). Thanks again.

-------------------------------------------------------------------------------------------------------------------

// See SetupX_Template.h for all options available
#define RPI_DISPLAY_TYPE
#define ILI9486_DRIVER // 20MHz maximum SPI

#define TFT_MISO 19
#define TFT_MOSI 23
#define TFT_SCLK 18
#define TFT_CS 15   // Chip select control pin
#define TFT_DC 2   // Data Command control pin
#define TFT_RST 4   // Reset pin (could connect to RST pin)
//#define TFT_RST  -1 // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST

#define TOUCH_CS 22     // Chip select pin (T_CS) of touch screen

#define LOAD_GLCD   
#define LOAD_FONT2 
#define LOAD_FONT4  
#define LOAD_FONT6 
#define LOAD_FONT7 
#define LOAD_FONT8  
#define LOAD_GFXFF  

#define SMOOTH_FONT
#define SPI_FREQUENCY  16000000 // Testing
#define SPI_TOUCH_FREQUENCY  2500000

Bodmer, 2020-04-22:
I have tried "Setup11_RPi_touch_ILI9486.h" on my ESP32 and RPi display and it works fine up to 20MHz SPI rate.

It could be that you display has a different controller chip so it would be worth patiently trying others such as ILI9481. If you are using jumpers on a breadboard or long wires then I would rewire and keep wires short. Make sure the display is getting 5V, some ESP32 boards do not provide 5V. Make sure the GND (0V) connection is reliable, if I use jumper wires I use two GND wires as this is the signal return path and is critical.

Other than that I have run out of ideas!

dimosap, 2020-05-12:
Hi calm4git, 
Thanks for your response. Yes, my display worked with a rpi (touch was working too).  My current user setup file looks like this:

#define RPI_DISPLAY_TYPE
#define ILI9486_DRIVER // 20MHz maximum SPI
#define TFT_MISO 35
#define TFT_MOSI 33
#define TFT_SCLK 32
#define TFT_CS   5  // Chip select control pin
#define TFT_DC    12  // Data Command control pin
//#define TFT_RST   4  // Reset pin (could connect to RST pin)
#define TFT_RST  -1  // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define TOUCH_CS 4     // Chip select pin (T_CS) of touch screen
#define SPI_FREQUENCY  2000000
#define SPI_TOUCH_FREQUENCY  2500000

With these settings only image works, no touch. Also tried pins 22,17,34,14 for TOUCH_CS pins with no success. I added a cap between EN and GND but it didn't yield any results. The project in the link you provided is the one i want to recreate too. Let me know if you find something that could solve my problem. Thanks in advance.

calm4git, 2020-05-12:
Hi dimosap,

For the touch, the question is, if the controller is still transmitting valid data, or transmitting anything at all. Please check what you get as raw values. Inside the library you have a function to read the raw data, may use a serial console for debugging purposes and see if the values are something useful.

Regads

CalM

dimosap, 2020-05-12:
Thanks for your quick response, just ran Test_Touch_Controller sketch. It seems that it's not transmitting anything. I get the same values over and over while touching the screen ( just in case, x:8191 y:8991 z:4095).

Bodmer, 2020-05-13:
Is the screen working OK now, so it is just touch problems left?

dimosap, 2020-05-16:
Hello again and thanks for your time spent on this issue. I finally made it work. It seems that when soldering the display i didn't notice that i shorted MISO and SCK by mistake hence only image displayed and touch wasn't working. However even before soldering, sketches such as ILI9486 and ILI9481 didn't match my display. The final header file worked for me was the one below. Thanks once again for the wonderful support!  

#define RPI_DISPLAY_TYPE
#define ILI9486_DRIVER // 20MHz maximum SPI
#define TFT_MISO 35
#define TFT_MOSI 33
#define TFT_SCLK 32
#define TFT_CS 5 // Chip select control pin
#define TFT_DC 12 // Data Command control pin
//#define TFT_RST 4 // Reset pin (could connect to RST pin)
#define TFT_RST -1 // Set TFT_RST to -1 if display RESET is connected to ESP32 board RST
#define TOUCH_CS 4 // Chip select pin (T_CS) of touch screen
#define SPI_FREQUENCY 2000000
#define SPI_TOUCH_FREQUENCY 2500000

closed 2020-06-06 by Bodmer