Bodmer/TFT_eSPI #606: When pushImage is used for 16-bit sprites, the effect of swap bytes is reversed.

lovyan03, 2020-04-24:
Sprite.cpp Lines 24: 
 _iswapBytes = false;   // Do not swap pushImage colour bytes by default

Here it says, If the swap byte is false, it will not swap.
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L24

Lines 847 and 953 perform a swap if the variable is true.
lines 832 and 937 perform a swap if the variable is False.

https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L832
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L847
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L937
https://github.com/Bodmer/TFT_eSPI/blob/master/Extensions/Sprite.cpp#L953

Perhaps the conditions on lines 832 and 937 need to be amended.

Bodmer, 2020-04-25:
Yes, this is deliberate. The Sprites pixel colors are stored as big endian whereas the processor is actually little endian.

If you ave an example sketch where this is causing an issue then please post a copy.

lovyan03, 2020-04-25:
Yes, I agree with that. The internal representation of the sprite should be big-endian so that it can be done with no transformation when doing a pushSprite.
So, whether drawing to a sprite or drawing to SPI, it's both big-endian. In other words, we think that setSwapBytes works in the same way.
Let's try it out.

```c
// Modify and use TFT_Flash_Bitmap in examples.
#include <TFT_eSPI.h>
#include "Info.h"

static TFT_eSPI tft = TFT_eSPI();
static TFT_eSprite sp (&tft);

void setup() {
  tft.begin();
  tft.fillScreen(TFT_BLACK);

  tft.setSwapBytes(true);
  tft.pushImage(0,  0, infoWidth, infoHeight, info);
  tft.setSwapBytes(false);
  tft.pushImage(0, 40, infoWidth, infoHeight, info);

  sp.createSprite(32, 32);
  sp.setSwapBytes(true);
  sp.pushImage(0, 0, infoWidth, infoHeight, info);
  sp.pushSprite(60, 0);
  sp.setSwapBytes(false);
  sp.pushImage(0, 0, infoWidth, infoHeight, info);
  sp.pushSprite(60, 40);
}
```

[image]

I tried pushImage the same bitmap for SPI and Sprite, with different conditions for setSwapBytes. Then, the conditions are reversed when drawing to SPI and when drawing to Sprite.
I'm assuming that the same bitmap will be drawn in the same way, whichever way it's drawn, as long as it has the same setSwapBytes condition, what do you think?

Bodmer, 2020-04-26:
Thanks for the example, this demonstrated and inconsistency that I thought I had eliminated. This is not fixed.  The changes break the Rotated_Sprite_3 sketch but this uses the TFT_eFEX library which I will need to update.

Bodmer, 2020-04-26:
Fixed.

commit 90af737f (2020-04-26): Fix #606 inconsistency + others

TFT_eFEX also needs updating so Rotated_Sprite_3 example renders correctly.

pushImage for FLASH images updated so partly off-screen images are correctly rendered.
closed 2020-04-26 by Bodmer