Bodmer/TFT_eSPI #609: 1bpp sprites vs. swapBytes

jpasqua, 2020-05-01:
I just updated to the latest version and the colors of my 1bpp sprites looked as through swapBytes was set incorrectly.

The display I use needs swapBytes set to true normally. All of my graphics (including 565 images) look good with this setting. However, when I use 1bpp sprites the colors I set using setBitmapColor() look like they have their bytes swapped.

I looked at the current code and the old code and it appears that the code path differs in their use of setSwapBytes(). 

To make my code work for drawing direct to the screen and for 1bpp sprites, I must setSwapBytes(true) normally, and then set it to false before performing a pushSprite(). Then I restore it of course.

I am also using 4bpp sprites with a cmap and those work fine.

Bodmer, 2020-05-01:
Thanks for pointing this out. I have updated the library so the sketch swapBytes setting does not affect 1, 4 and 8 bit pushSprites as they have a fixed format.

closed 2020-05-01 by Bodmer

jpasqua, 2020-05-02:
Thanks for the quick fix.