Bodmer/TFT_eSPI #616: SPIFFS.h no such file while compiling.

ghosttk, 2020-05-09:
Compiling examples for esp32 module, I get this error in TFT_eSPI_ESP32.h
There should be "SPIFFS.h" in the Processors directory.
Where to find this file?

#ifdef SMOOTH_FONT
  // Call up the SPIFFS (SPI FLASH Filing System) for the anti-aliased fonts
  #define FS_NO_GLOBALS
  #include <FS.h>
  #include "SPIFFS.h" // ESP32 only
  #define FONT_FS_AVAILABLE
#endif

Bodmer, 2020-05-09:
SPIFFS.h is part of the Arduino ESP32 board package, make sure that is loaded.

ghosttk, 2020-05-10:
Currently I can not fully  ESP32 board package nor PlatformIO libs because of the network firewall.
Can TFT_eSPI be used in esp_idf ?

Bodmer, 2020-05-10:
The library needs the Arduino boards package as it uses many of the Arduino specific function calls.

closed 2020-05-10 by Bodmer