Bodmer/TFT_eSPI #618: 'TFT_MOSI' / 'TFT_SCLK' was not declared

cyberman54, 2020-05-09:
Since version upgrade 2.2.3 -> 2.2.4 i run on a compiler error, if i include the library without having all SPI pins defined.

> Compiling .pio\build\usb\lib5d4\TFT_eSPI_ID1559\TFT_eSPI.cpp.o
> In file included from .pio\libdeps\usb\TFT_eSPI_ID1559\TFT_eSPI.cpp:20:0:
> .pio\libdeps\usb\TFT_eSPI_ID1559\Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA()':
> .pio\libdeps\usb\TFT_eSPI_ID1559\Processors/TFT_eSPI_ESP32.c:675:20: error: 'TFT_MOSI' was not declared in this scope     
>      .mosi_io_num = TFT_MOSI,
>                     ^
> .pio\libdeps\usb\TFT_eSPI_ID1559\Processors/TFT_eSPI_ESP32.c:677:20: error: 'TFT_SCLK' was not declared in this scope     
>      .sclk_io_num = TFT_SCLK,
>                     ^
> *** [.pio\build\usb\lib5d4\TFT_eSPI_ID1559\TFT_eSPI.cpp.o] Error 1

I will post a pull request with a solution.

Bodmer, 2020-05-09:
Your setup has probably been over-written so make sure you have defined the pins for an ESP32.

cyberman54, 2020-05-09:
I'm maintaining a multi hardware platform software. That means your lib may be included, but not used. In this case pins are undefined.

Bodmer, 2020-05-09:
The library will not run correctly on an ESP32 with SPI interface unless these pins are defined.
Define some dummy values and include in User_Setup.h

Bodmer, 2020-05-13:
I have updated the library so that ESP SPI pins can be left undefined by a user and ensure SDA read functions will continue to work.

closed 2020-05-13 by Bodmer

cyberman54, 2020-05-15:
@Bodmer I tested with v2.2.6, but still getting compiler error :-(

> In file included from .pio\libdeps\usb\TFT_eSPI_ID1559\TFT_eSPI.cpp:20:0:
> .pio\libdeps\usb\TFT_eSPI_ID1559\Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA()':
> .pio\libdeps\usb\TFT_eSPI_ID1559\Processors/TFT_eSPI_ESP32.c:666:20: error: 'TFT_MOSI' was not declared in this scope
>      .mosi_io_num = TFT_MOSI,
>                     ^
> .pio\libdeps\usb\TFT_eSPI_ID1559\Processors/TFT_eSPI_ESP32.c:668:20: error: 'TFT_SCLK' was not declared in this scope
>      .sclk_io_num = TFT_SCLK,
>                     ^
> *** [.pio\build\usb\lib5d4\TFT_eSPI_ID1559\TFT_eSPI.cpp.o] Error 1

cyberman54, 2020-05-15:
@Bodmer oops, i just figured out that this patch is not yet released. Tested with HEAD, works. Thanks.

martinglass, 2020-07-07:
I'm also getting the same error with v2.2.6, using Arduino compiler with the ESP32 Dev Module board option. I can't figure out where to enter the correct values. Can you help?
Here's the error log:
  In file included from ~/Documents/Arduino/libraries/TFT_eSPI/TFT_eSPI.cpp:20:0:
~/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c: In member function 'bool TFT_eSPI::initDMA()':
~/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:666:20: error: 'TFT_MOSI' was not declared in this scope
     .mosi_io_num = TFT_MOSI,
                    ^
~/Documents/Arduino/libraries/TFT_eSPI/Processors/TFT_eSPI_ESP32.c:668:20: error: 'TFT_SCLK' was not declared in this scope
     .sclk_io_num = TFT_SCLK,
                    ^
exit status 1
Error compiling for board ESP32 Dev Module.

QuationSong, 2022-07-26:
You can define a new number, like this:
#define TFT_CS   PIN_D6  // Chip select control pin D8
#define TFT_DC   PIN_D9  // Data Command control pin
#define TFT_RST  PIN_D7  // Reset pin (could connect to NodeMCU RST, see next line)
#define TFT_SCLK -1
#define TFT_MOSI -1