Bodmer/TFT_eSPI #620: Display fails with wifi enabled

theologyandstuff, 2020-05-09:
So I'm trying to do this terminal-esque project with an esp32 and an ST7789V display. It all works fine until I bring WiFi into the mix. When I uncomment the WiFi.begin line below, the display shows nothing. It's not a time-based thing... I've tested with delays and it doesn't fail when it connects to wifi, it fails with any compile where the wifi.begin line is uncommented. Also, it seems that the file size only jumps when I uncomment that line, not when I include the WiFi.h file. Here is my source below:

```
#include <TFT_eSPI.h>
#include <WiFi.h>
#include <WiFiUdp.h>

#define DEVICE_NAME "MASS_ESP32_1"
#define APP_NAME "M.A.S.S."

void UpdateTFT(void *pvParameters);

TFT_eSPI tft = TFT_eSPI();
TFT_eSprite buf = TFT_eSprite(&tft);

char* ssid      = "ORBI64";
char* password  = "lewiswireless";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  delay(10);
  setupTFT();
  centerMessage(F("INITIALIZING"));
  pushBuffer();
  
//  WiFi.disconnect(true);
//  WiFi.onEvent(WiFiEvent);

//  this is the incriminating line:
  WiFi.begin(ssid, password);
  
  xTaskCreate(UpdateTFT, "UpdateTFT", 2048, NULL, 1, NULL);
}

void loop() {
  
}

void setupTFT() {
  tft.init();
  tft.setRotation(0);
  clearScreen();
  tft.setTextColor(TFT_WHITE);

  buf.createSprite(240, 240);
  buf.setColorDepth(1);
  clearBuffer();
  pushBuffer();
}

void UpdateTFT(void *pvParameters) {
  (void) pvParameters;

  while (true) {
    drawRootDeviceMeta();
    drawWiFiStatus();
    pushBuffer();
    delay(100);
//    Serial.println("looping");
  }
}

void drawRootDeviceMeta() {
  buf.setTextColor(TFT_WHITE);
  buf.drawString(F(APP_NAME), 0, 0, 4);
//  buf.setTextColor(TFT_GREEN);
  buf.drawString(F(DEVICE_NAME), 26, 20, 2);
}

[... 307 lines cut ...]
// The XPT2046 requires a lower SPI clock rate of 2.5MHz so we define that here:
#define SPI_TOUCH_FREQUENCY  2500000

// The ESP32 has 2 free SPI ports i.e. VSPI and HSPI, the VSPI is the default.
// If the VSPI port is in use and pins are not accessible (e.g. TTGO T-Beam)
// then uncomment the following line:
#define USE_HSPI_PORT

// Comment out the following #define if "SPI Transactions" do not need to be
// supported. When commented out the code size will be smaller and sketches will
// run slightly faster, so leave it commented out unless you need it!

// Transaction support is needed to work with SD library but not needed with TFT_SdFat
// Transaction support is required if other SPI devices are connected.

// Transactions are automatically enabled by the library for an ESP32 (to use HAL mutex)
// so changing it here has no effect

#define SUPPORT_TRANSACTIONS

```

I've tried different pins, different wifi channels etc. I don't know if it's an SPI problem caused by Wifi, or some software problem like memory. Also, when I say the display "fails", I mean that when I reboot the board the display turns on, displays random garbage for a split second like normal, then displays black pixels. Now that I think about, I haven't tried not using the buffer sprite. Maybe that's the issue.

Thanks in advance for your help, and also thanks for creating this awesome library!

Bodmer, 2020-05-09:
I am surprised it works at all...

You need to specify the color depth before creating the Sprite as the default depth is 16 bits:
```
  buf.setColorDepth(1);
  buf.createSprite(240, 240);
```
Otherwise you are trying to grab a huge chunk of memory 240*240*2 = 115200 bytes (more than is actually available). See if that works.

P.S. I deleted your SSID and password from your post.

theologyandstuff, 2020-05-09:
Ah okay I'll try that

theologyandstuff, 2020-05-09:
Works perfectly, thanks! That was it!

closed 2020-05-09 by theologyandstuff