Bodmer/TFT_eSPI #623: 'TFT_BACKLIGHT_ON' was not declared in this scope

happy12, 2020-05-12:
Arduino IDE: 1.8.12
hardware: ESP32
Windows 10

I just upgraded to the new release of this library and now the code does not compile anymore. I get this error:
'TFT_BACKLIGHT_ON' was not declared in this scope

Bodmer, 2020-05-13:
For what line number is the error reported? I suspect the line is in your sketch.

Check you have defined the backlight settings is the setup file. Bear in mind that upgrading the library will over-write changes to the example setup files.

closed 2020-05-24 by Bodmer

vablech, 2020-06-06:
Hi Bodmer,

I run into this also.

The complete message is

//
/Arduino/Test/Test.ino: In function 'void setup()':
Test:156:30: error: 'TFT_BACKLIGHT_ON' was not declared in this scope
         digitalWrite(TFT_BL, TFT_BACKLIGHT_ON); // Turn backlight on. TFT_BACKLIGHT_ON has been set in the TFT_eSPI library in the User Setup file TTGO_T_Display.h
                              ^
exit status 1
'TFT_BACKLIGHT_ON' was not declared in this scope
//

Cheers

Ralph

Bodmer, 2020-06-06:
@vablech
Are you using the Arduino IDE?

Note: defining the TFT I/O pins in the sketch will not work with the Arduino IDE.

vablech, 2020-06-06:
Hi Bodmer,

yes, I'm using the arduino IDE (fresh installed from Fedora 31 Repo with updates) including TFT_eSPI version 2.2.6 as is provided through the arduino ide.

The Testcode itself is that one https://github.com/Xinyuan-LilyGO/TTGO-T-Display/blob/master/TTGO-T-Display.ino.
That's what they put on the ESP during production before shipping - and it was working sometime before to compile ;)

cheers

Ralph

vablech, 2020-06-06:
just ... needless to mention that it has no impact on compiling, what they did with peripherals around the ESP32 and how they are connected - this is a different story - but at least it should compile.

Bodmer, 2020-06-06:
I suspect you have a rogue copy of the TFT_eSPI library somewhere in your system as the TFT_BACKLIGHT_ON is clearly [defined here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup25_TTGO_T_Display.h#L22).  I tried the test code and it compiles and runs fine.

Note that TTGO copied the TFT_eSPI library into their own version of the library and I do not support that copy.

vablech, 2020-06-06:
the library is the one that is rolled out via the arduino ide linked libs.

That's what is in the Setup25_TTGO_Display.h:

//
#define TFT_BL          4  // Display backlight control pin
#define TFT_BACKLIGHT_ON HIGH  // HIGH or LOW are options
//
[Setup25_TTGO_T_Display.h.txt](https://github.com/Bodmer/TFT_eSPI/files/4740644/Setup25_TTGO_T_Display.h.txt)
in ./Arduino/libraries/TFT_eSPI/User_Setups/Setup25_TTGO_T_Display.h

Bodmer, 2020-06-06:
Check again you have followed the [instructions here](https://github.com/Xinyuan-LilyGO/TTGO-T-Display). In particular ALL the edits described in section 2.

vablech, 2020-06-06:
Hi Bodmer,

ouch - yes, you're correct. Classic RTFM error on my side.

Sorry that I have wasted your time with that.

The good thing is, that whenever s/o will run into that, they will hopefully google and find this thread.

I followed that :  https://sites.google.com/site/jmaathuis/arduino/lilygo-ttgo-t-display-esp32
To be honest, I was like TLDR :-(

Thanks for your help

Ralph