Bodmer/TFT_eSPI #624: tft and sprites color mismatches

athusrv, 2020-05-12:
Hi. I've realized this little color mismatch with the same color being shown in the TFT area (without using sprite) and a sprite area. I believe this is related to the color depth, which I can understand. I'm using a sprite with an 8bit color depth. 16bit color depth produces the same color with no issues but I can't use it due to memory limitations.

I'd like to understand if there is a workaround so I can have a match of this color in both areas.
I was thinking about some alternatives:

- Adjust the gamma which requires getting the right gamma value for it.
- Setting the "normal" area  (no sprite) to use 8bit color depth as well (is that possible? if so, how can I do that?)

For reference see this image below. At the top is the TFT "normal" area and at the bottom is the sprite area.

[image]

Bodmer, 2020-05-13:
You are right, this is because color resolution is lost when converting to 8 bit colors and back to 16.

The simplest was to fix this is to write a function that takes the 16 bit color you want as a TFT background, converts to 8 bits and then convert back to 16 again. The background color will then match the Sprite color. There are functions to help. Example:

tft.fillScreen( tft.color8to16( tft.color16to8( TFT_SKYBLUE ) ) );

closed 2020-05-13 by Bodmer

athusrv, 2020-05-13:
Thanks! That works!