Bodmer/TFT_eSPI #625: readRect(), pushImage(), setSwapBytes() compatibility

jpasqua, 2020-05-12:
During long running operations such as hitting a REST API on a remote server, I provide a visual indication by overlaying an info icon on the screen. Before doing so I read the pixels under the icon position so I can write them back later. I do this using readRect(). I then display the icon and later, when the long operation is over, I replace the icon with the pixels I read earlier.

I have used both pushRect (which is deprecated) and pushImage() to push the pixels back. Both result in pixels exhibiting the swapped bytes effect. My display normally has swpaBytes == true. To work around this I set swapBytes to false, perform the pushImage, then restore swapBytes to true.

Bodmer, 2020-05-13:
I have updated the lbrary to reinstate pushRect() for use, so it can be used as with the partner function readREct() and so setting swapBytes to false will not be needed.

Thanks for raising this.

closed 2020-05-13 by Bodmer

jpasqua, 2020-05-13:
Works great. Thank you for the quick update.

jpasqua, 2020-05-28:
@bodmer, if you'd like to see the project I'm building with your library, check out the [repo](https://github.com/jpasqua/MultiMon) or this [video](https://www.youtube.com/watch?v=MYjuCYKBxmM)

commit 1314a34c (2020-05-13): Fix #625

pushRect() reinstated to pair with readRect() so swapBytes does not need to be changed from true to false before calling.