Bodmer/TFT_eSPI #628: Request: Font 6: Characters to display temperature

ArnieO, 2020-05-13:
It would be great to use Font 6 to display temperatures. 

I propose:

- Do as you did for Font 2: Replace the grave (`) diacritical mark (character 0x60) with the degree (°) symbol. 
- Add 'F' and 'C' characters
- Add '+' and '-'

Bodmer, 2020-05-13:
I will add this to my list of potential enhancements.

The font was originally generated about 6 years ago and I can't find the code to generate new characters in the required Run Length Encoded format so I may have to reinvent the wheel and start again...

The approach I use is to add the degree symbol and F/C as superscripts in a smaller font, for example:
[image]

ArnieO, 2020-05-13:
Thank you for your rapid response!
No problem, I can do this as a workaround.

closed 2020-05-13 by Bodmer