Bodmer/TFT_eSPI #629: Compile errors like   'xset_mask' was not declared in this scope!

ghosttk, 2020-05-14:
I  had change settings in User_setup.h, using default 9341 driver with 8 bits parallel board.
#define TFT_PARALLEL_8_BIT

But got  many compilel errors , most of them  was the   'xset_mask' was not declared.
I checked the Processors/TFT_eSPI_ESP32.h, founding it was in there.

`
#if defined (TFT_PARALLEL_8_BIT)

  // Create a bit set lookup table for data bus - wastes 1kbyte of RAM but speeds things up dramatically
  // can then use e.g. GPIO.out_w1ts = set_mask(0xFF); to set data bus to 0xFF
  #define CONSTRUCTOR_INIT_TFT_DATA_BUS            \
  for (int32_t c = 0; c<256; c++)                  \
  {                                                \
    xset_mask[c] = 0;                              \
    if ( c & 0x01 ) xset_mask[c] |= (1 << TFT_D0); \
    if ( c & 0x02 ) xset_mask[c] |= (1 << TFT_D1); \
    if ( c & 0x04 ) xset_mask[c] |= (1 << TFT_D2); \
    if ( c & 0x08 ) xset_mask[c] |= (1 << TFT_D3); \
    if ( c & 0x10 ) xset_mask[c] |= (1 << TFT_D4); \
    if ( c & 0x20 ) xset_mask[c] |= (1 << TFT_D5); \
    if ( c & 0x40 ) xset_mask[c] |= (1 << TFT_D6); \
    if ( c & 0x80 ) xset_mask[c] |= (1 << TFT_D7); \
  }                                                \

  // Mask for the 8 data bits to set pin directions
  #define dir_mask ((1 << TFT_D0) | (1 << TFT_D1) | (1 << TFT_D2) | (1 << TFT_D3) | (1 << TFT_D4) | (1 << TFT_D5) | (1 << TFT_D6) | (1 << TFT_D7))

  // Data bits and the write line are cleared to 0 in one step
  #define clr_mask (dir_mask | (1 << TFT_WR))

  // A lookup table is used to set the different bit patterns, this uses 1kByte of RAM
  #define set_mask(C) xset_mask[C] // 63fps Sprite rendering test 33% faster, graphicstest only 1.8% faster than shifting in real time
`

[errors.txt](https://github.com/Bodmer/TFT_eSPI/files/4627156/errors.txt)

[User_Setup.h.txt](https://github.com/Bodmer/TFT_eSPI/files/4627161/User_Setup.h.txt)

Bodmer, 2020-05-14:
As a temporary fix, add:
#define ESP32_PARALLEL
To your setup file.

Also you must comment out all the ESP8266 pin defines in your setup file.

ghosttk, 2020-05-14:
#define ESP32_PARALLEL works, thankyou.

Bodmer, 2020-05-14:
Thanks for pointing out this issue. It is now fixed on Github.

closed 2020-05-14 by Bodmer