Bodmer/TFT_eSPI #630: ESP32 SPI DMA - when enabled SPI reads from TFT/Touch controller do not work

Bodmer, 2020-05-14:
When DMA is enabled via initDMA() reads from the SPI port no longer work. deInitDMA() does not correct this either.

Reason unknown.
Fix not known.

Work around: Do not use DMA on ESP32 unless reads from thye SPI bus are not needed.

fvanroie, 2020-11-15:
I ran into this issue today, where DMA was working without touch but not with touch enabled.
Disabling DMA indeed can be used as a workaround... but not ideal performance wise.

Bodmer, 2020-11-15:
Thanks for raising this. I have fixed this.  Here is my test code:

```
// This sketch is to test the touch controller, nothing is displayed
// on the TFT.  The TFT_eSPI library must be configured to suit your
// pins used. Make sure both the touch chip select and the TFT chip
// select are correctly defined to avoid SPI bus contention.

// Make sure you have defined a pin for the touch controller chip
// select line in the user setup file or you will see "no member"
// compile errors for the touch functions!

// It is a support and diagnostic sketch for the TFT_eSPI library:
// https://github.com/Bodmer/TFT_eSPI

// The "raw" (unprocessed) touch sensor outputs are sent to the
// serial port. Touching the screen should show changes to the x, y
// and z values. x and y are raw ADC readings, not pixel coordinates.

#include <SPI.h>
#include <TFT_eSPI.h>
TFT_eSPI tft = TFT_eSPI();
TFT_eSprite img = TFT_eSprite(&tft);  // Sprite class
uint16_t * ptr;

//====================================================================

void setup(void) {
  Serial.begin(115200);
  Serial.println("\n\nStarting...");

  tft.init();
  tft.initDMA();
  tft.fillScreen(TFT_RED);
  ptr = (uint16_t*)img.createSprite(10, 10);
}

//====================================================================

void loop() {

  img.fillSprite(random(0x10000));
  tft.startWrite();
  tft.pushImageDMA(random(tft.width()), random(tft.height()), 10, 10, ptr);
  tft.dmaWait();
  tft.endWrite();

  uint16_t x, y;

  tft.getTouchRaw(&x, &y);

  Serial.printf("x: %i     ", x);

  Serial.printf("y: %i     ", y);

  Serial.printf("z: %i \n", tft.getTouchRawZ());

  delay(250);

}

//====================================================================

```

closed 2020-11-15 by Bodmer

fvanroie, 2020-11-15:
Wow, that is just amazing. I have just tested it and works fine now!
Thanks so much for your quick reply and fix!

commit 579961bd (2020-11-15): Fix #630