Bodmer/TFT_eSPI #631: White screen using ESP32 8-Bit Parallel

SilentPotat0, 2020-05-15:
recently got a mikroe-495 tft display with ili9341 for a project, i hooked it up and editted the setup file. but i only end up with a white display no matter what, all the IM pins are ground so it should be setup for 8-bit parallel. as a last hope i also tried using SPI but still the same results no matter what example i use. im using a cheap 30 pin ESP32 from china, but that hasnt been a problem for me before. also tried connecting external power to it but no difference.
the setup file
  `// See SetupX_Template.h for all options available

#define ESP32_PARALLEL

#define ILI9341_DRIVER

// The ESP32 and TFT the pins used for testing are:
#define TFT_CS   33  // Chip select control pin (library pulls permanently low
#define TFT_DC   14  // Data Command control pin - must use a pin in the range 0-31
#define TFT_RST  32  // Reset pin, toggles on startup

#define TFT_WR    15  // Write strobe control pin - must use a pin in the range 0-31
#define TFT_RD    27  // Read strobe control pin

#define TFT_D0   12  // Must use pins in the range 0-31 for the data bus
#define TFT_D1   13  // so a single register write sets/clears all bits.
#define TFT_D2   26  // Pins can be randomly assigned, this does not affect
#define TFT_D3   25  // TFT screen update performance.
#define TFT_D4   5
#define TFT_D5   17
#define TFT_D6   16
#define TFT_D7   4

#define LOAD_GLCD   // Font 1. Original Adafruit 8 pixel font needs ~1820 bytes in FLASH
#define LOAD_FONT2  // Font 2. Small 16 pixel high font, needs ~3534 bytes in FLASH, 96 characters
#define LOAD_FONT4  // Font 4. Medium 26 pixel high font, needs ~5848 bytes in FLASH, 96 characters
#define LOAD_FONT6  // Font 6. Large 48 pixel font, needs ~2666 bytes in FLASH, only characters 1234567890:-.apm
#define LOAD_FONT7  // Font 7. 7 segment 48 pixel font, needs ~2438 bytes in FLASH, only characters 1234567890:.
#define LOAD_FONT8  // Font 8. Large 75 pixel font needs ~3256 bytes in FLASH, only characters 1234567890:-.
#define LOAD_GFXFF  // FreeFonts. Include access to the 48 Adafruit_GFX free fonts FF1 to FF48 and custom fonts

#define SMOOTH_FONT
`

Bodmer, 2020-05-16:
The library does work with the ILI9341 in 8 bit parallel and also with SPI mode. The SPI mode must be 4 wire.

Check the setup you expect is being picked up by the compiler by running the "Read_User_Setup" test sketch. For parallel mode check your connections are short and you have a good ground connection. If you are using wires then keep them short  (<100mm) as the ESP32 outputs fast edges that can cause ground bounce. Link several GND (0V) wires to help reduce GND noise.

LongLiveCHIEF, 2020-05-25:
I echo @Bodmer 's comments.  I get white screen a lot when testing on a breadboard because the breadboard connections aren't super secure. If I go through and make sure each and every one is tight, then reset, it'll usually bring the program up.

I'm at the point where I've started to draw up a custom PCB breakout that I'm going to mill it out on my cnc just to eliminate all those pesky connection issues.

SilentPotat0, 2020-05-25:
might be sadly im in the middle of exam season so dont have the time to solder it up, saw several people having problems with breadboards so it isnt unlikely, will test it as soon as i can, but its gonna be a bit.

Bodmer, 2020-05-31:
Open another issue when your have investigated further and if you have not found the solution.

closed 2020-05-31 by Bodmer