Bodmer/TFT_eSPI #632: functions not in scope

LongLiveCHIEF, 2020-05-17:
I've got a fresh platformio project, and the first thing I've added is TFT_eSPI.  My `main.cpp` is a direct copy/paste of the `READ_USER_SETUP` example.

When I build i get `was not declared in this scope` errors for `printProcessorName`, `getPinName` and `lationn`.

When I try the `TFT_meters` example, i get even more `was not declared in this scope` errors for just about every function used.

Here's a link to this first commit, and when built will reproduce errors: https://github.com/LongLiveCHIEF/benchtop_psu/tree/e6a92bcce8809b8479249a72282ca599e01ef185

I've also tried it using `-include` build flag instead of the `-D` vars, with the same result.

LongLiveCHIEF, 2020-05-17:
I'm starting to think this is a platformio error with how it's executing the gcc compiler.

LongLiveCHIEF, 2020-05-17:
n/m... it's a forward declaration thing.  Fixed with reordering definitions.

closed 2020-05-17 by LongLiveCHIEF

Bodmer, 2020-05-19:
If you let me know which order worked then I will update the example.