Bodmer/TFT_eSPI #633: Screen flickering .. using Font_Demo_1

schimkat, 2020-05-18:
Hi Bodmer .. and everyone else. :-)

I'm struggling with screen flickering, which appears every time objects are on top of each other. Right now I'm testing the Font_Demo_1 (SPIFFS), and the problem appears it this demo, as well as with my own code.

For reproducing, I just alter line 152 (in Font_Demo_1), for displaying the same string every time, which helps showing the issue:

Line 152:     tft.print("Test");

When printing the same string each time, it's easy to see that flickering appears in the area, that has been "erased" by the fillRect method on line 149.

I'm using the 2.2" inch display (http://tiny.cc/ztsapz) on a nodemcu board.

Any hints anyone? :-)

schimkat, 2020-05-18:
Found the solution .. wrap everything in sprites. :-)

Bodmer, 2020-05-19:
Well done was about to suggest that.

closed 2020-05-19 by Bodmer