Bodmer/TFT_eSPI #635: LCD with ILI9488 + touchscreen + SC card on ESP32's VSPI do not work

DJ534, 2020-05-20:
Hallo Bodmer,
first I admire you work for the community. The library is great! Thank you for providing it and for its regular maintenance too!

I connected the TFT LCD with ILI9488 chip, its touchscreen and its SD card slot to ESP32's VSPI. MISO of the LCD was left unconnected and the IRQ pin too. All other pins were connected to ESP32. I managed to run the LCD and touchscreen simultaneously on the VSPI together with SD card slot on the HSPI. This means, that I have successfully run the test and diagnostic sketches for the LCD and touchscreen. With a minor change I have run SD_test on HSPI (see code below). But when I hook them up all together to VSPI, the SD_Test sketch (see below for VSPI) returns: Card Mount Failed. The modifications of the SD_Test I have made is the following:

For VSPI:
```

SPIClass SDSPI(VSPI);

const uint8_t SDss = 27;
const int8_t Vsck = 18;
const int8_t Vmiso = 19;// misto GPIO 12 - nesmi byt HIGH behem bootu
const int8_t Vmosi = 23;

void setup(){
    Serial.begin(115200);
    
    SDSPI.begin( Vsck, Vmiso, Vmosi, SDss);
    
    if(!SD.begin( SDss, SDSPI)){
        Serial.println("Card Mount Failed");
        return;
    }

// rest of the code for SD_Test sketch ....

}

```

For HSPI:

```

SPIClass SDSPI(HSPI);

const uint8_t SDss = 27;
const int8_t Hsck = 14;
const int8_t Hmiso = 26;// misto GPIO 12 - nesmi byt HIGH behem bootu
const int8_t Hmosi = 13;

void setup(){
    Serial.begin(115200);
    
    SDSPI.begin( Hsck, Hmiso, Hmosi, SDss);
    
    if(!SD.begin( SDss, SDSPI)){
        Serial.println("Card Mount Failed");
        return;
    }

// rest of the code for SD_Test sketch ....

```

I have the following questions:

1. Is it possible to make LCD with ILI9488, its touchscreen and SD card reader run on VSPI? If yes, where am I making a mistake?

2. Is it possible to hook up LCD's touchscreen and LCD with ILI9488 (with MISO connected) on one SPI?

3. When I run Test_touch_controller I am continuously obtaining readings from one point on the edge of the screen. The z coordinate is small - to 30 or 40. What is its cause? 

Remark: I miss comprehensive recommended pin connection for ESP32 in the User_Setup.h This is mentioned for ESP 8266, but not for ESP 32. What I am missing are these lines:

```

// We must use hardware SPI, a minimum of 3 GPIO pins is needed.
// Typical setup for ESP8266 NodeMCU ESP-12 is :
//
// Display SDO/MISO  to NodeMCU pin D6 (or leave disconnected if not reading TFT)
// Display LED       to NodeMCU pin VIN (or 5V, see below)
// Display SCK       to NodeMCU pin D5
// Display SDI/MOSI  to NodeMCU pin D7
// Display DC (RS/AO)to NodeMCU pin D3
// Display RESET     to NodeMCU pin D4 (or RST, see below)
// Display CS        to NodeMCU pin D8 (or GND, see below)
// Display GND       to NodeMCU pin GND (0V)
// Display VCC       to NodeMCU 5V or 3.3V

```
for ESP32.

Thanks for help.

Bodmer, 2020-05-20:
If you have multiple devices on the SPI bud (TFT, touch, SD) then if you just run a sketch for one of them,say the SD card then you must drive the chip selects high for all the other devices  right at the start of setup() otherwise multiple devices can end up driving the bus at the same time.

DJ534, 2020-05-21:
That makes sense. Thanks. But from my point of view, that does not answer the three questions in my comment. Can you answer them please? They are just below the first block of code.

Bodmer, 2020-05-21:
1. Yes, see previous comment.
2. No, the TFT MISO does not tri-state. However you might find it works if you add a 1K series resistor in the wire from the TFT MISO pin to the ESP82666, so oter devices on the bus can still drive the bus.
3. Perhaps there was another device on the SPI bus that had not been disabled? Maybe you had TFT MISO wired in?

DJ534, 2020-05-21:
Thanks for the answers!

closed 2020-05-24 by Bodmer