Bodmer/TFT_eSPI #638: Screen artifacts when using ST7796S SPI display with ESP32

ernicek, 2020-05-22:
Hello,

I have an interesting issue - I'm trying to use 4" display driven by ST7796S connected to ESP32 with this "user setup" file:
```
#define ST7796_DRIVER
#define TFT_CS   17 
#define TFT_DC   16
//#define TFT_RST  PIN_D4 
#define TFT_RST  5 
#define TFT_SCLK 18
#define TFT_MISO 19
#define TFT_MOSI 23
#define TOUCH_CS 33
#define SPI_FREQUENCY  40000000
```
 I tried several demos from your library and it works perfectly, without any problem. So I move on - I want to use a "GUI" - so I'm using LittlevGL/lv_arduino library for that. And when I tried the basic demo from LittlevGL I saw small lines on the display... you can see them on the attached video orig.mov. I don't know if you know this library, but in short - there is a small function (I know it could be written more efficiently - but I want to keep it simple) - which simply copies pixels from LittlevGL buffer to your "driver" with using 'tft.writeColor' function:
```
    uint16_t c = LV_COLOR_SILVER.full;
    tft.startWrite(); /* Start new TFT transaction */
    tft.setAddrWindow(area->x1, area->y1, (area->x2 - area->x1 + 1), (area->y2 - area->y1 + 1)); /* set the working window */
    for (int y = area->y1; y <= area->y2; y++) {
      for (int x = area->x1; x <= area->x2; x++) {
        c = color_p->full; <------  comment this line
        tft.writeColor(c, 1);
        color_p++;
     }
  }
  tft.endWrite(); /* terminate TFT transaction */
```

I made an experiment - I commented out the line where 'c' variable is getting a new color from 'color_p' pointer, so 'c' variable contains still the same color silver (initialized on the beginning of the code).  You can see the result in the second video file - modified.mov - the small, almost black lines are still there...

I also used this almost the same "settings" with another display driven by ILI9488 or ILI9341 it works without any issue...

Is this is a known issue? Did I make something wrong?

[videos.zip](https://github.com/Bodmer/TFT_eSPI/files/4669896/videos.zip)

Bodmer, 2020-05-23:
I suspect there is an interrupt occurring that is associated with the touch function which is using the SPI bus while a screen update is in progress.
tft.startWrite sets the screen chip select low until tft.endWrite is called so is not thread safe. In that case you need to fix the issue in the touch handler.

ernicek, 2020-05-29:
Thank you for your answer, but unfortunately - I'm still facing the same problem. Yes, to be honest I'm not so good in this HW area, and this is what I tried:
As the touch handler, I'm using the one included in your library - so, I just periodically call "tft.getTouch(&X, &Y, 600)" and that's it. So, first I tried to use your private member 'inTransaction' for a check it inside 'getTouch' function, but without success. I also tried to use 'enterCritical' and 'exitCritical' around the 'startWrite' & 'endWrite' section, but it was also without success...

And again - what is the difference between this one ST7796S and maybe ILI9488,... because these works well, they are just small for my needs :-)

Bodmer, 2020-05-31:
Since you have had success with the ILI9488 and ILI9341 then I am not sure where the problem lies.

Have you tried reducing the SPI clock rate?

closed 2020-06-06 by Bodmer

andrewleek, 2023-02-09:
I was having the same problem with the same ST7796S SPI display (my display is actually v1.1) and ESP32.  I tried altering the SPI clock rate, but the artifacts continued.  In the end this is what worked for me: https://github.com/Bodmer/TFT_eSPI/discussions/898
- Please note that on the v1.1 board the D1 diode is in a different location but the solution is the same.

I hope this helps someone encountering this issue.

Thanks for this amazing library!

Cheers!