Bodmer/TFT_eSPI #639: Add support for choosing BGR or RGB order on ST7735 enhancement

pkoevesdi, 2020-05-24:
I have ST7735 160x128 Display. My display has Red and Blue swapped.
The option TFT_RGB_ORDER in User_Setup.h has no influence.
Is there a workaround, until (hopefully) the option works for this Display too?

Thank You!

pkoevesdi, 2020-05-24:
I was able to get the expected behaviour with adding
```
    tft.writecommand(TFT_MADCTL);
    tft.writedata(B11001100);
```
in front of my script. Thus, the script is not hardware agnostic anymore.

pkoevesdi, 2020-05-24:
...but my workaround seems to get overwritten by any
`tft.setRotation(#)`
I think I understand, that tft.setRotation writes to the same register. But it doesn't keep the other bits set there, i.e. 03 für RGB order - so resets them back.

Bodmer, 2020-05-24:
These ST7725 displays come in a number of variants. The display setup used to be able to be determined by the color if the tab on the screen protection film but this is no longer the case. However the different variant options remain. I suspect that choosing a different color tab in the setup will correct the color swap.

In the longer term I will add making the RGB/BGR setting work for all displays to my future enhancement list.

Bodmer, 2020-05-24:
Setting options are[ here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setup.h#L86-L94).

pkoevesdi, 2020-05-24:
Sorry, I forgot to tell, that I tried all of them. The only one, that has no tray pixels at the edges is ST7735_GREENTAB2, but this (and most or all of the others too) have swapped colors.

Bodmer, 2020-11-02:
Added support for choosing BGR or RGB order on ST7735 in version 2.3.4

closed 2020-11-02 by Bodmer