Bodmer/TFT_eSPI #640: Error compiling All_Free_Fonts_Demo in platfromio - Free_Fonts.h: no such file or directory

LongLiveCHIEF, 2020-05-25:
_Note: Opening this here for now, but as I dig in it may wind up being moved to platformio._

***
When running the `All_Free_Fonts_Demo`  using the platformio build, I get the following compile time error

<details>
<summary>Toggle to view log output</summary>  

```txt
Processing nodemcu32s (board: nodemcu-32s; platform: espressif32; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/nodemcu-32s.html
PLATFORM: Espressif 32 1.11.1 > NodeMCU-32S
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 2.10004.191002 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 29 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <TFT_eSPI> 2.2.8
|   |-- <SPIFFS> 1.0
|   |   |-- <FS> 1.0
|   |-- <FS> 1.0
|   |-- <SPI> 1.0
|-- <SPI> 1.0
Building in release mode
Compiling .pio/build/nodemcu32s/src/main.cpp.o
Generating partitions .pio/build/nodemcu32s/partitions.bin
Compiling .pio/build/nodemcu32s/lib1e9/FS/FS.cpp.o
Compiling .pio/build/nodemcu32s/lib1e9/FS/vfs_api.cpp.o
src/main.cpp:336:75: fatal error: Free_Fonts.h: No such file or directory

********************************************************************
* Looking for Free_Fonts.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Free_Fonts.h"
* Web  > https://platformio.org/lib/search?query=header:Free_Fonts.h
*
********************************************************************

compilation terminated.
```
</details>

<details>
<summary>Here's my `platformio.ini`</summary>

```ini
[platformio]
description = "Firmware for a smart DIY Benchtop Power Supply"
default_envs = nodemcu32s

[env]
platform = espressif32
framework = arduino
lib_deps = TFT_eSPI

[env:nodemcu32s]
board = nodemcu-32s
build_flags =
  -Os
  -D USER_SETUP_LOADED=1
  -D ILI9341_DRIVER=1
  -D TFT_WIDTH=240
  -D TFT_HEIGHT=320
  -D TFT_MISO=19
  -D TFT_MOSI=23
  -D TFT_SCLK=18
  -D TFT_CS=15
  -D TFT_DC=2
  -D TFT_RST=4
  -D LOAD_GFXFF=1
  -D LOAD_GLCD=1
  -D SPI_FREQUENCY=24000000
```
</details>

I'm getting a similar error when i use Adafruit GFX library, but for `header:Wire.h`.  

When looking at the [TFT_eSPI headers recognized by platfromio](https://platformio.org/lib/show/1559/TFT_eSPI/headers) , it clearly shows `Free_Fonts.h` in the headers list, which indicates this is likely a compiler error.

At the moment, I believe this has something to do with pre-compiler instructions in the source, in combination with using `include "some_header.h"` as oppossed to `include <some_header.h>`, which causes the compiler to act differently.

I'm going to dig, and will either submit a PR if it's an easy fix, a detailed findings if it's not, or move it to platform.io if it's something that _should_ be found based on include paths given to the compiler by platformio.

LongLiveCHIEF, 2020-05-25:
Ran `find . -name Free_Fonts.h` and got this:

```
$ find . -name Free_Fonts.h
./examples/320 x 240/All_Free_Fonts_Demo/Free_Fonts.h
./examples/320 x 240/Free_Font_Demo/Free_Fonts.h
./examples/480 x 320/Free_Font_Demo/Free_Fonts.h
```

Next, I added `-include $PROJECT_LIBDEPS_DIR/$PIOENV/TFT_eSPI_ID1559/examples/320\ x\ 240/All_Free_Fonts_Demo/Free_Fonts.h` to my `platformio.ini`, and that resulted in the following error during compilation: 

```
Compiling .pio/build/nodemcu32s/src/main.cpp.o
cc1plus: fatal error: home/chief/code/benchtop_psu/.pio/libdeps/nodemcu32s/TFT_eSPI_ID1559/examples/320 x 240/All_Free_Fonts_Demo/Free_Fonts.h: No such file or directory`
```

So I got to thinking that the compiler is having problems because it is finding 3 different `Free_Fonts.h` headers no matter what we do.

To start testing this theory, i switched the precompiler isntructions in my `main.cpp` back to `#include "Free_Fonts.h"` so it would only look in the current directory, and copied the `Free_Fonts.h` from the relevant examples folder into that directory... and compilation worked.

There are a few approaches to test to see what is easiest for the user if it is in fact a multiple match situation, but in the meantime, I figured I would take a gander at what the differences were in the 3 different header files all named `Free_Fonts.h`.

I used [diff3](https://linux.die.net/man/1/diff3) to analyze, and the differences are minimal, and seem to be based on a font inclusion matrix based on screen size.

```
diff3 examples/320\ x\ 240/All_Free_Fonts_Demo/Free_Fonts.h examples/320\ x\ 240/All_Free_Fonts_Demo/Free_Fonts.h examples/480\ x\ 320/Free_Font_Demo/Free_Fonts.h
====3
1:42c
2:42c
  #define LOAD_GFXFF
3:41a
====3
1:46a
2:46a
3:46c
  // Call up the font using tft.setTextFont()
====3
1:55c
2:55c
  // Use the following when calling setFont()
3:55c
  // Use the following when calling tft.setFreeFont()
====3
1:266a
2:266a
3:267,268c
  #define TT1 1
```

There are multiple paths to fix this, but not sure what approach is preferred by @Bodmer .  Perhaps the _easiest_ approach would be to give a unique header name to each of the headers.  The best approach might be to bring this header file into the main source and use a definition to include/exclude certain fonts based on precompiler instructions?

Seems like there's already some of that going on based on the way fonts are included from `TFT_eSPI.h` (which includes `Fonts/GFXFF/gfxfont.h`).

Makes me wonder if the `Free_Fonts.h` is still needed at all?

LongLiveCHIEF, 2020-05-25:
@Bodmer maybe it would be simpler and easier to maintain if we created a precompiler instruction that allows you to define a minimum supported height/width resolution in each font definition header file, and then we could get rid of all the duplicate code in the examples folder for the various resolutions, and just have a `examples/<example_name>/<example_src>` pattern for the examples folder instead?

Bodmer, 2020-05-25:
I think the pre-compiler options would be a problem for the Arduino IDE environment because it places all examples automatically into the IDE examples menu. I would expect Platform IO with the Arduino plugin to handle the examples, but I have little experience with PlatformIO so this may not be correct.

Bodmer, 2020-05-31:
I will close as I have no experience with PlatformIO...

closed 2020-05-31 by Bodmer