Bodmer/TFT_eSPI #642: DMA for ILI9488

momosh13, 2020-05-27:
Please add DMA for ILI9488/ESP32.
Thank you very much for this library.

Bodmer, 2020-05-31:
The ESP32 DMA mode is for SPI displays only. Unfortunately the ILI9488 only works in 18 bits per pixel mode with a SPI interface. This means that all the native format 16 bit pixels pixels will need to be converted 18 bits per pixel in 3 bytes before using DMA. The perforamnce an memory penalty will be significant and negate the small DMA benefit.

Depending on your needs you may be able to use the second processor in the ESP32 to handle rendering or background processing. [See simple example here.](https://github.com/Bodmer/TJpg_Decoder/blob/master/examples/ESP32_Dual_Core_Flash_Jpg/ESP32_Dual_Core_Flash_Jpg.ino)

closed 2020-05-31 by Bodmer

kiddingme1, 2020-08-23:
Hi Bodmer, 

I was looking through the ili9488 data sheet and it indicates that 4 wire SPI it supports 3, 16 and 18 bits per pixel.  3 wire only supports 3 and 18 bits per pixel.

> The available display data formats are:
> 	 8 colors, RGB 1, 1, 1 bits input (set Standard Command 3Ah, DBI [2:0] as 001)
> 	 65K-Colors, RGB 5, 6, 5 bits input data (set Standard Command 3Ah, DBI [2:0] as 101)
> 	 262K-Colors, RGB 6, 6, 6 bits input data (set Standard Command 3Ah, DBI [2:0] as 110)

Is there a way to support DMA for 4 wire SPI with your library? 

I have the two following setups: 

ESP32 with 4 wire SPI ili9488
ESP32 with 8 bit parallel ili9488

The ili9488 datasheet indicates both of those interfaces support 16 bit per pixel. Is there anyway to get your library to support DMA with these interfaces? 

Thank you for all of your hard work.

Trackhe, 2020-12-04:
I m also Interested in

Bodmer, 2020-12-04:
Sadly, despite what it states in the data sheet the ILI9481/6/8 displays do not work in 16 bit colour mode when the 4 wire SPI interface is used.

If you wish to use DMA then the simplest option is to use a SPI ST7796 display which I find works reliably at 80MHz SPI clock speed.

kiddingme1, 2020-12-04:
For the ili9488 I ended up bit banging 16bit parallel interface using a ESP32-S2. Nice thing is you can use GPIO’s 0-15 to send a whole 16bit pixel at once without any bit shifting. Unfortunately the bit shifting is a relatively small time savings versus the time it takes to toggle the GPIO port. In order to send a 16bit word it takes a minimum of three writes to the GPIO port(clr bits and WR, set bits, and toggle WR high) which on the ESP32(and S2) is very slow. Obviously still faster then 8bit parallel but I wonder if the 80mhz plus DMA of the ST7796 Bodmer mentioned would be overall even better.

kiddingme1, 2020-12-04:
If you don’t care about overwriting GPIO 16-31(again talking about ESP32-S2) you can reduce each pixel write from three GPIO port writes down to two for a decent performance bump.

Bodmer, 2020-12-04:
You can test any display combination on a bare ESP32 (for example DMA ST7796 80MHz SPI) with a sketch to check performances. This is possible because the library does not rely on a display being connected.

kiddingme1, 2020-12-05:
Oh wow, never realized that. Thanks. I never tried to integrate the 16bit parallel ESP32-S2 code with your library but I can measure screen fill time pretty easily for quick comparison. 

ILI9488 SPI 40MHz:     ~100ms/10fps
ILI9488 8 Bit Parallel:   ~40ms/25fps
ST7796 SPI 80MHz:     ~36ms/28fps
ILI9488 16 Bit Parallel: ~23ms/43fps

With that being said, while running LVGL demo's and scrolling up and down the screen I cant visually see much difference between the 8 bit parallel and 16 bit parallel interfaces. The ILI9488 SPI 40MHz is pretty bad though. For my project I settled on the ESP32 with 8 bit parallel interface because I did not want to give up the dual core. Looking forward to learning more about the ESP32-S3...sounds like it may have the high GPIO count like the S2 but still have dual core like the ESP32

j45p41, 2021-02-12:
Hi @kiddingme1 and @Bodmer i have posted this[](https://forum.lvgl.io/t/esp32-ili9488-8-bit-parallel-choppy-animation-when-scrolling-tabs-with-lvgl-generated-objects/4540/14)  issue on the lvgl forum. I am using ER-TFT035-6 display which uses the ili9488 driver connected using 8 but parallel and running TFT_eSPI. I am being pointed to the choppiness being due to the driver however I would be interested to see if you have any thoughts on how to improve performance.

kiddingme1, 2021-02-12:
> Hi @kiddingme1 and @Bodmer i have posted [this](https://forum.lvgl.io/t/esp32-ili9488-8-bit-parallel-choppy-animation-when-scrolling-tabs-with-lvgl-generated-objects/4540/12) issue on the lvgl forum. I am using ER-TFT035-6 display which uses the ili9488 driver connected using 8 bit parallel and running TFT_eSPI on esp32. I am being pointed to the choppiness being due to the driver however I would be interested to see if you have any thoughts on how to improve performance.

Bodmer’s 8-bit driver is as optimized as possible. Unfortunately the ESP32 GPIO port write speed is relatively slow at ~50ns. Hoping newer ESP32 variants will be faster but that is one of the reasons others have achieved higher frame rates with STM modules. I achieved higher frame rates with a ESP32-s2 in 16-bit parallel mode but it wasn’t a noticeable difference to the eye. Are you able to run the LVGL demo’s and share a video for comparison? I have them running on mine and they do scroll considerably smoother then your video.

unre4l, 2021-03-26:
@kiddingme1 Hi, this is slightly off-topic, but i just stumbelt across your comment. 
I am also trying to run this lib on the esp32s2 but cant work my head around the spi conversion.

To use esp32s2's `SPI3` i added:
```
#define VSPI 1
#define VSPI_HOST 3
```
but TFT_eSPI requires also definitions for `VSPID_OUT_IDX` and `VSPIQ_IN_IDX`. 
Do you mind providing your sketch or User_Setup.h?

kiddingme1, 2021-03-26:
I never tried to use Bodmers library with the s2 so not sure I ran into same issues as you. I bought the s2 specifically for the higher GPIO count to enable 16bit parallel interface with the ili9488. Wrote custom 16bit driver. Never had a need to mess with the spi on the s2. 

Sorry!