Bodmer/TFT_eSPI #643: JPG to BMP

mrstanlez, 2020-05-27:
Hello Mr. Bodmer,

please can you make a simple function for convert JPG to BMP file from SD card to SD card ?
Thank you.

kamorris, 2020-05-28:
Hello Mr Stanlez,  It's not so simple to write code to convert jpg to bmp files; looking at https://en.wikipedia.org/wiki/JPEG, the file format is more complex than bmp.  I generally use gimp (which is free) to open the files and save as bmp format.  There are other solutions for converting bmp files to arrays if needed.

mrstanlez, 2020-05-28:
Hello Kate Morris,
thank you for answering my question. But I wish to do with ESP32, without
any other computer or software. Only read jpeg from SD card and convert it,
save to the SD card.

With java I can write PC program within  5 minute.

Thank you in advance.

kamorris, 2020-05-30:
Hello Mr Stanlez,

Have you looked at https://github.com/Bodmer/TJpg_Decoder ?

mrstanlez, 2020-05-30:
Hello Kate,
yes, but I dont know how to get rgb from each pixel from the picture.
So I decided to create own function for converting jpg2bmp wirh esp32.

Thank you. Have a nice day.

kamorris, 2020-05-30:
Sounds like you have resolved your issue then?

Bodmer, 2020-05-31:
If you are using the TJpg_Decoder library then you can get the 24 bit colors output during the decoding [by changing this line](https://github.com/Bodmer/TJpg_Decoder/blob/master/src/tjpgd.h#L10) to:
```
#define JD_FORMAT       0
```
You then get 3 bytes per pixel (R+G+B).

closed 2020-06-06 by Bodmer