Bodmer/TFT_eSPI #644: Question? Drawstring/float beyond screen edge

tablatronix, 2020-05-28:
How can I push this float closer to the bottom edge, it is a float and has no descenders, I assume that is why I cannot. Also I cannot seem to pass a y larger than the screen width(height) to push it down..
[image]

Bodmer, 2020-05-31:
This is a historical limitation of the font rendering approach for the Run Length Encoded fonts. The library automatically pulls the character back to the screen area otherwise the character image gets corrupted. The code could be changed but it would then render characters more slowly, This font rendering approach was developed many years ago for the UNO/Mega to improve font rendering performance. There is no simple solution unless Sprites are used.  Smooth fonts (stored in FLASH rather than in SPIFFS to improve performance) will work though there will be a performance penalty and you will get flickering unless Sprites are used.

closed 2020-06-06 by Bodmer

Bodmer, 2020-06-11:
I have updated the library to allow partially off screen RLE font rendering. (now version 2.2.11)