Bodmer/TFT_eSPI #645: Flickering when redrawing freefonts

tablatronix, 2020-05-30:
If I use your font I get no redraw flickering at all.

```C++
 tft.setTextColor(WHITE,BLACK);
 tft.drawFloat((float)str,1,lpad,TFT_WIDTH,4); 
```

If i use a freefont, It flickers and I can see it scanning, tearing
```C++
 tft.setTextColor(WHITE,BLACK);
 tft.setFreeFont(AA_FONT_MED); 
 tft.drawFloat((float)str,1,lpad,TFT_WIDTH,GFXFF); 
```

Why are these rendering differently? Do your fonts somehow buffer the changes an dnot have to redraw the padding constantly or something ?

I have video.

tablatronix, 2020-05-30:
Top tears, bottom does not
[image]

tablatronix, 2020-05-30:
Ok so these fonts are bliting windowed multiple lines at once? and gfx are not using blocks but redrawing the entire padding rect ?

tablatronix, 2020-05-30:
I wound up rewriting it all using sprites and NotoSansBold36, which is working ok. I was afraid they would be slower but its not bad.

Still curious how this works or how to make this easier, eg. auto creating and wrapping freefonts in sprites or block writes or why they cannot be optimized the same way

tablatronix, 2020-05-30:
Actually frame rate is a little better now, I wonder what the performance would be with full buffer..

tablatronix, 2020-05-30:
Holy crap, yeah so I think I just need to make more espi fonts for the sizes I need..

Not definitive but my use case.

// sprite 55fps/66fps varies by font
// gfx 72fps  - freefont()
// block font 88fps - loadFont()

Bodmer, 2020-05-31:
This is just down to the different rendering approaches. Adafruit method does not add background while the glyph is being rendered whereas the build in fonts do. This means that the entire text area has to be blanked first when rendering the Adafruit font. I suspect Adafruit chose this simple approach as itallic glyphs may overlap the previous+next character zones, so this complicates rendering with a background. It would be possible to develop a smarter rendering function but I don't like the Adafruit font format, so never added this.  The larger built in fonts are Run Length Encoded as they were developed many years ago for a Mega based project that had more limited FLASH. Sadly, due to a PC upgrade a few years ago since I seem to have lost the font generator code for the RLE fonts...

closed 2020-05-31 by Bodmer

tablatronix, 2020-05-31:
Thanks, do you have any documentation on the RLE fonts ? The code looks pretty straight forward but was curious if it was better documented somewhere.

I like the gfx format in that is is bitmap standard and compatible with most tools, you can also use it to layer glyphs, and reuse lcd icons and stuff from other lcd projects.

Bodmer, 2020-05-31:
In the RLE font a character is drawn as a raster scan within a box size of the characters width and height. The RLE format is a byte, the top bit is a flag, the least significant 7 bits are a 7 bit number (0-127). If the flag is 1 then foreground pixels are plotted, if it is 0 the background pixels are plotted. The number of pixels to plot is the 7 bit number +1.

So, for example, the encoded sequence "0x81, 0x1F" means plot 2 pixels in foreground then 32 pixels in background.

I have been meaning to write  Processing sketch to generate new RLE fonts but have not started it yet.