Bodmer/TFT_eSPI #646: Compiler warning: SPIFFS deprecated

jpasqua, 2020-05-31:
Arduino IDE: 1.8.10
ESP8266 Core: 2.7.1

While compiler with all warnings enabled:
````
In file included from ...,
                 from ...:
/.../Arduino/libraries/TFT_eSPI/Extensions/Smooth_font.h:47:23: warning: 'SPIFFS' is deprecated (declared at /.../Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.1/cores/esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
   fs::FS   &fontFS  = SPIFFS;
````

Bodmer, 2020-05-31:
Thanks, I will update the library at some point. I had been waiting for LittleFS to be added to the board package but I see this has now been done.

Incidentally I have already added code for LittleFS to the TJpg_Decoder library but this is commented out and completely untested!

SPIFFS will however remain in the library as the ESP32 has [not adopted LittleFS yet](https://github.com/espressif/arduino-esp32/issues/3765).

jpasqua, 2020-05-31:
Thank you. I was just cleaning up warnings in my own project and ran into this. It is obviously not a problem - I just wanted to bring it to your attention.

Bodmer, 2020-11-14:
LittleFS Smooth Font examples have been added. LittleFS is significantly faster than SPIFFs when used with the ESP8266 and Smooth Fonts.

SPIFFS will be retained for the ESP32.

closed 2020-11-14 by Bodmer

spiff72, 2020-11-29:
Hello, I am not sure if I am asking in the correct place, as this isn't the PIO library, but I am trying to compile WLED using a custom usermod (I created it and submitted it to WLED in a pull request about 9 months ago.  It is specifically for using WLED with a TTGO T-display to show info about the current settings (palette, color, etc) on the inbuilt display of the TTGO ESP32 module.  It all worked back then, but for some reason, I can't get it to compile with the latest version of WLED cloned from github.  I get an error message as follows:

`In file included from .pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.h:52:0,
                 from .pio\libdeps\esp32dev\TFT_eSPI\TFT_eSPI.cpp:17:
.pio\libdeps\esp32dev\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:68:36: fatal error: SPIFFS.h: No such file or directory

****************************************************************
* Looking for SPIFFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPIFFS.h"
* Web  > https://platformio.org/lib/search?query=header:SPIFFS.h
*
****************************************************************`
I tried going to the link shown above, but it finds no results.  So I was hoping you could answer this question - Where is the spiffs.h header file supposed to come from?

Thanks!

Bodmer, 2020-11-29:
SPIFFS.h is part of the Arduino board package for the ESP32.

spiff72, 2020-11-29:
I just caught my error.  I forgot to copy the usermod file that I had created into the correct folder.  This is what I get for not looking at this for 9 months and trying to recall how to make it all work.

It just compiled successfully now that I moved that file to the right place.

Thanks!